evereux/pycatia #1: Installation Error

ogroll, 2019-05-21:
When trying to install pycatia step by step with the installation guide, I had the following Error: ERROR: Could not install packages due to an EnvironmentError: No such file or directory: 'C:\\...\\AppData\\Local\\Temp\\pip-install-mldm2ljt\\pycatia\\pycatia-0.1.4.dist-info\\WHEEL'

I tried it with two versions of python, which where the latest python version of v3.7.3 and version 3.6.8 with the same result.

Is there anything I can do to resolve this issue?

Greets, Oliver

evereux, 2019-05-21:
Hello Oliver,

Yep, just tried myself and have the same issue which I'll try and get fixed now if I can.

In the meantime you can download the zip from github and place the pycatia folder in your working folder: https://github.com/evereux/pycatia/archive/master.zip

I'll post back if and when I've solved the issue...

Thanks for reporting!

Paul

evereux, 2019-05-21:
Should work now but you'll have to clear the pip cache. On my machine I deleted everything from the folder: `C:\Users\<username>\AppData\Local\pip`.

I don't know what the problem was but I just cleared out all my pypi build and dist files, rebuilt and resent to pypi.

Thanks again.

ogroll, 2019-05-21:
Hi Paul,

it works, awesome. Thank you for your quick reply and your solution. I cleared the pip cache and installed the package without any problems.

Cheers

closed 2019-05-21 by ogroll