evereux/pycatia #2: class：HybridShapeFactory, how do i  draw a line?

Elemt, 2019-07-19:
How do I draw a line between two points

evereux, 2019-09-20:
Hi Elemt,

I was planning on providing an example but I've just not had time.

Whilst it is possible with pycatia it's currently out of scope of the pycatias built in functions. To do it would require a good understanding of the underlying CATIA COM API and to use that. Though would not really require pycatia as I'm sure you could do that without it.

The question itself is rather open though ... do we assume we already have them selected within the script?

evereux, 2020-06-15:
This is now possible with latest version.

Take a look at `example_015.py`. 

This creates a new part, creates two points and creates a line between them.

evereux, 2020-07-21:
I'm closing this issue as I've provided an example and there has been no feedback.

closed 2020-07-21 by evereux