evereux/pycatia #3: How to export STEP file from Surf using pycatia package?

selvakarna, 2019-10-03:
How to export STEP file from Surf using pycatia package?

evereux, 2019-10-05:
This isn't currently directly available using pycatia but it can be done by accessing the Document com object. See the file V5Automation.chm within your CATIA installation for more information.

`from pycatia import CATIAApplication

# path to file to open.
file_name = r'Part1.CATPart'

catia = CATIAApplication()
# open document
documents = catia.documents()
documents.open(file_name)

document = catia.document()
# export data to new format.
document.document.ExportData(r"C:\Users\paul\pycatia\exported_stp", "stp")
# close document
document.close()`

evereux, 2019-10-06:
I'm an idiot. I did add this feature.

    document.document.ExportData(r"C:\Users\evereux\pycatia\exported_stp", "stp")

can be

    document.export_data(r"C:\Users\evereux\pycatia\exported_stp", "stp")

You can use the documents.open() method to open other file types.

selvakarna, 2019-10-09:
Hi @evereux , 

when i install pycatia in my Linux machine ,i got error below, how to solve this

ERROR: No matching distribution found for pywin32==224 (from pycatia)

selvakarna, 2019-10-10:
i am asking surface to step file? @evereux

evereux, 2019-10-10:
Do you have CATIA V5 running in Linux? I know it's possible but it's technically a Windows application as is pywin32.

evereux, 2020-01-23:
I'm closing this ticket as I've no feedback from OP.

closed 2020-01-23 by evereux