evereux/pycatia #6: No attribute Open in catia.documents ? how to read CATPart?

selvakarna, 2019-10-11:
doc.open('1.CATPart')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'function' object has no attribute 'open':

evereux, 2019-10-13:
Can you please respond to my question to you here: https://github.com/evereux/pycatia/issues/3.

Also, there are several examples in the documentatio that include the opening of a .CATPart which you aren't following. I have also provided an example in the issue linked above which is nothing like what you are trying to do here.

You can't open a CATPart without even at least importing the module you are raising a ticket on.

selvakarna, 2019-10-14:
Hi @evereux , you dont know how to solve this issue ????????????? okay thanks

closed 2019-10-14 by selvakarna