evereux/pycatia #9: How to get the positon matrix via python?

Macusliu, 2019-10-27:
It is easy to set the position, however, it cannot get the position matrix. Is it feasible?

evereux, 2019-10-27:
The position matrix of what?

 Example #2 shows how to do that for points: https://pypi.org/project/pycatia/

evereux, 2019-10-29:
I saw an email in my spam folder asking about getting position matrix of a product. I'm guessing this came from you?

    It is glad to see the greate module "pycatia 0.1.5.1". I am a tranditional C++ programmer focused on functional developing of CATIA. However it's impossible to get the position matrix of a certain item in the product tree via this module. So is there any ways that can get the matrix? Hoping for your response.

I've not yet added that feature but I'm looking at it now.

evereux, 2019-10-29:
Version 0.1.6 now has that feature.

You can see a usage example here: https://pycatia.readthedocs.io/en/latest/examples.html#example-9

closed 2019-12-23 by evereux