evereux/pycatia #10: Obtain Bounding Box from components in a Product

mrjapon, 2019-11-13:
Hello,
I'm trying to obtain the Bounding Box of several ítems in a Product. The Bounding Box is available once is launched the 'Measure Inertia' inside CATIA, and clicking on Principal Axis in Customize options (see https://www.cadcam-group.eu/catia-v5-bounding-box). Once you have done this, it is easy to recover these parameters from the CATIA tree.
The problem is that I cannot automatize 'Measure Inertia', that is what measures and save the parameters in the tree.
I've seen you have some examples that obtain the CoG, mass, and some other parameters. Is it posible to do so to get these three parameter of the Bounding Box?
Thank you

evereux, 2019-11-13:
Hi,

I know exactly what you're talking about but haven't looked at automating
this. The problem I see is that the bounding box can differ whether the
root is selected or individual body/bodies. Making it select the root in
all cases is potentially dangerous.

It's certainly possible I think but off the top of my head I can't think of
a nice fool proof way to implement ... unless the automation fires off many
warnings.

Paul

mrjapon, 2019-11-13:
Thk for your answer. In fact, I was thinking about the bounding box (and
cog) of subproducts inside a product, so that problem would not exist,
since each root-subproduct  is the item to measure

evereux, 2019-11-13:
Aah ok got you. I've never used the feature like that. Let me have a look at the automation API and see if I can come up with something. Might be a few days but I'll definitely respond.

mrjapon, 2019-11-14:
This macro (in VB) can help for sure. I'm studying and translating it to Python. Let's see....
https://www.eng-tips.com/viewthread.cfm?qid=436208

evereux, 2020-01-23:
Sorry I've not responded sooner.

I've looked at the VB script and whilst it's certainly doable in python it's not something I currently have time to look at in detail sorry. Maybe in the near future.

tcrakshi, 2020-07-19:
How to get the user defined properties in part or product?

i am  trying with "user_ref_properties", its throwing me a com  error like below.

com_error: (-2147352567, 'Exception occurred.', (0, 'CATIAProduct', 'The method UserRefProperties failed', None, 0, -2147467259), None)

Can any one help me in that?

evereux, 2020-07-19:
@tcrakshi  This is unrelated to the original post. A new issue has been raised. I will reply to the issue there.

manjitkumar, 2020-09-07:
@mrjapon I see you faced this issue some time back. Were you able to make any progress on this?

mrjapon, 2020-09-14:
Sorry, I left this issue stopped. One day I'll take it back and find the
code for it. Good luck!!

Tian-Jionglu, 2020-09-25:
Hi,
I think the key of this issue is getting a correct Inertia Object.
With pycatia 0.4.1, it can basically work.
```
# test Inertia

from pycatia import catia
from pycatia.space_analyses_interfaces.inertia import Inertia

caa = catia()

document = caa.active_document
product = document.product()

inertia_com = product.get_technological_object('Inertia')
my_inertia = Inertia(inertia_com)
print(my_ineria.density)
print(my_inertia.mass)
```
The `density` and `mass` properties can be obtained correctly. Methods like `get_cog_position` or `get_principal_axes` yet doesn't work.
I'm trying to solve this puzzle...

evereux, 2020-09-26:
Try this:

```python
from pycatia import catia

caa = catia()

document = caa.active_document
product = document.product()

spa_workbench = document.spa_workbench()
inertias = spa_workbench.inertias

# clear existing inertias.
# this is required since the CAA object seems to keep the inertia collection in
# memory on each iteration of the script. Not sure if this is by design or 
# something pycatia should be doing.
for i in range(inertias.count):
    inertias.remove(1)

# add product to inertia collection.
inertias.add(product)

for inertia in inertias:
    print(inertia.mass)
    print(inertia.get_cog_position())```

evereux, 2020-09-26:
> The `density` and `mass` properties can be obtained correctly. Methods like `get_cog_position` or `get_principal_axes` yet doesn't work.
> I'm trying to solve this puzzle...

Good luck! I've been trying to solve this and can't figure out the problem.

Tian-Jionglu, 2020-09-27:
Aha, that's the correct solution to get a exact Inertia instance. But it's pity I didn't find the bound box parameters directly.

I tried another solution, simply and directly translated from 

> This macro (in VB) can help for sure. I'm studying and translating it to Python. Let's see....
https://www.eng-tips.com/viewthread.cfm?qid=436208

```
from pycatia import catia

caa = catia()

document = caa.active_document
product = document.product()

sel= document.selection

sel.clear
sel.add(product)
caa.start_command("Measure Inertia")

print(f'BBLx: {product.parameters.item("BBLx").value_as_string()}',
     f'BBLy: {product.parameters.item("BBLy").value_as_string()}',
     f'BBLz: {product.parameters.item("BBLz").value_as_string()}',
     sep='\n')
```
Here is one more issue should we pay attention, as CAA Document describe StartCommand:

> Role:This method starts a command and executes it untill its first interaction. Please notice interactions such as selections you could add after in your macro will not work. StartCommand is useful to execute one-shot (not interactive) commands, it is not safe for interactive commands.

I think "Measure Inertia" is not one-shot command. To solve this, it would be anther issue #39.

@mrjapon, please check whether it is you mean.

evereux, 2020-09-27:
If I understand correct and you want the script to stop and wait for input from a user could you use a [messagebox](https://github.com/evereux/pycatia/blob/e5802bf68d7afa145d4fd5ae295f781413c7f6ee/pycatia/in_interfaces/application.py#L1106) to prompt the user?

evereux, 2020-09-28:
This is actually more complex than I thought to make it interactive, for a number of reasons:

- You can't make a selection whilst the message box is open. So my suggestion above isn't valid.
- You can use pythons inbuilt `input()` to halt the process, select the product, enter to continue.
  - Having to switch between windows isn't user friendly.
  - You still need to close / cancel the inertia tool once the data is read. This isn't insurmountable as `pywinauto` could do this I think (see example 18, not directly related but uses `pywinauto`). However, there is a bigger problem to solve ... see next.
- In a large product `product.parameters` returns ALL parameters for ALL loaded documents.  `parameters.item('BBLx')` will only return the first occurrence.
  - Well, we could loop through the parameter names and get those that include the product name and the last BBLx no? Not really, as this is way too impractical for large products as it's too slow from my tests.

This is going to need some more thought.

Tian-Jionglu, 2020-09-29:
I've tried `win32gui.FindWindow`, `win32gui.CloseWind`, `win32gui.DestroyWindow`.
"Measure Inertia" Window can be find and close, but meet error when try to destroy. More work need to be done.

evereux, 2020-09-29:
This seems to work but comes with a warning. Please test thoroughly.

```python

"""

    Example 26:

    Prompt the user to select a product and get it's bounding box parameters
    
    .. warning:
        Currently there must be NO other existing Measure Inertias saved 
        ANYWHERE in your product tree as these may be returned and not 
        product you have selected.

"""

import win32con
import win32gui
from pycatia import catia

def close_inertia_window():
    # for future debugging from https://stackoverflow.com/questions/55547940/how-to-get-a-list-of-the-name-of-every-open-window
    # def winEnumHandler(hwnd, ctx):
    #     if win32gui.IsWindowVisible(hwnd):
    #         print(hex(hwnd), win32gui.GetWindowText(hwnd))
    #
    # win32gui.EnumWindows(winEnumHandler, None)

    handle = win32gui.FindWindow(None, "Measure Inertia")
    win32gui.PostMessage(handle, win32con.WM_CLOSE, 0, 0)

caa = catia()
document = caa.active_document
product = document.product()
selection = document.selection
selection.clear()

c = True
while c is True:
    input("Selection product to measure.\nPress <ENTER> when selection made.")
    selection = document.selection

    caa.start_command("Measure Inertia")
    parameters = product.parameters
    print(f"BBOx = {parameters.item('BBOx').value_as_string()}.")
    print(f"BBOy = {parameters.item('BBOy').value_as_string()}.")
    print(f"BBOz = {parameters.item('BBOz').value_as_string()}.")
    print(f"BBLx = {parameters.item('BBLx').value_as_string()}.")
    print(f"BBLy = {parameters.item('BBLy').value_as_string()}.")
    print(f"BBLz = {parameters.item('BBLz').value_as_string()}.")
    selection.clear()
    close_inertia_window()

    prompt = input("Continue? (Y/N):")

    if prompt.lower()[0] == 'n':
        c = False
    else:
        c = True
```

Tian-Jionglu, 2020-09-30:
👍 Brilliant!

tcrakshi, 2020-09-30:
It's really good example everex,
I think it will ans for #39 issue. 

def close_inertia_window(),  this function is closing command dog box, 
so now I want understand  is it similarly give input like yes/no/enter options, or recognise which all are button available in command dialog box and click on those button?

If it possible , thee we can address most of the issue, like generate catpart,. , Etc

Tian-Jionglu, 2020-09-30:
> so now I want understand is it similarly give input like yes/no/enter options, or recognise which all are button available in command dialog box and click on those button?
> 
> If it possible , thee we can address most of the issue, like generate catpart,. , Etc

It's possible.
However, it is done by pywin32, not pycatia. In this example, you can find it is win32gui play the role to close the window.
You can find more information by searching win32gui on Internet.

evereux, 2020-11-08:
> It's really good example everex,
> I think it will ans for #39 issue.
> 
> def close_inertia_window(), this function is closing command dialog box,
> so now I want understand is it similarly give input like yes/no/enter options, or recognise which all are button available in command dialog box and click on those button?
> 
> If it possible , thee we can address most of the issue, like generate catpart,. , Etc

See example 20. [message_box](https://pycatia.readthedocs.io/en/latest/api/pycatia/in_interfaces/application.html#pycatia.in_interfaces.application.Application.message_box) does what I think you're asking.

evereux, 2020-12-27:
Closing this as issue as I think it's resolved to a degree although the solution isn't perfect yet.

closed 2020-12-27 by evereux