evereux/pycatia #11: Can't install by using piptools

wyfhust, 2019-12-23:
error message:
Collecting pywin32==224 (from pycatia)
  ERROR: Could not find a version that satisfies the requirement pywin32==224 (from pycatia) (from versions: 225, 226, 227)
ERROR: No matching distribution found for pywin32==224 (from pycatia)

And install the specific pywin32 version(pip install pywin32==224)  also gives an error message, do I need to downgrade the pywin32.

evereux, 2019-12-23:
What operating system are you using?

wyfhust, 2019-12-23:
@evereux My os version is windows10 1903, and the catia version is v5r20.

evereux, 2019-12-23:
Which version of python?

evereux, 2019-12-23:
I've just tested Python 3.8.1 and had the same issue. There isn't a pypi package of pywin32 for this version of python.

You can get a wheel and install that from here: https://www.lfd.uci.edu/~gohlke/pythonlibs/#pywin32

wyfhust, 2019-12-24:
Thanks for the reply. Already installed pywin32 provided by the url and still got the same error message. It looks like that pycatia reqiure pywin32==224. Maybe I should INSTALL python3.7 and give a try. 

`C:\Users\R>pip install pywin32-227-cp38-cp38-win_amd64.whl
Requirement already satisfied: pywin32==227 from file:///C:/Users/R/pywin32-227-cp38-cp38-win_amd64.whl in c:\python38\lib\site-packages (227)

C:\Users\R>pip install pycatia
Collecting pycatia
  Using cached https://files.pythonhosted.org/packages/85/7d/21cce2c280beacb8c1bd1fcfc0a7eeb71f0dc02cbed83e7368680e595180/pycatia-0.1.6-py3-none-any.whl
Collecting attrs==18.2.0
  Using cached https://files.pythonhosted.org/packages/3a/e1/5f9023cc983f1a628a8c2fd051ad19e76ff7b142a0faf329336f9a62a514/attrs-18.2.0-py2.py3-none-any.whl
Collecting colorama==0.4.0
  Using cached https://files.pythonhosted.org/packages/0a/93/6e8289231675d561d476d656c2ee3a868c1cca207e16c118d4503b25e2bf/colorama-0.4.0-py2.py3-none-any.whl
Collecting py==1.7.0
  Using cached https://files.pythonhosted.org/packages/3e/c7/3da685ef117d42ac8d71af525208759742dd235f8094221fdaafcd3dba8f/py-1.7.0-py2.py3-none-any.whl
ERROR: Could not find a version that satisfies the requirement pywin32==224 (from pycatia) (from versions: 225, 226, 227)
ERROR: No matching distribution found for pywin32==224 (from pycatia)
`

wyfhust, 2019-12-24:
Thanks for the reply. Already installed pywin32 provided by the url and still got the same error message. It looks like that pycatia require pywin32==224. Maybe I should INSTALL python3.7 and give a try. 

`C:\Users\R>pip install pywin32-227-cp38-cp38-win_amd64.whl
Requirement already satisfied: pywin32==227 from file:///C:/Users/R/pywin32-227-cp38-cp38-win_amd64.whl in c:\python38\lib\site-packages (227)

C:\Users\R>pip install pycatia
Collecting pycatia
  Using cached https://files.pythonhosted.org/packages/85/7d/21cce2c280beacb8c1bd1fcfc0a7eeb71f0dc02cbed83e7368680e595180/pycatia-0.1.6-py3-none-any.whl
Collecting attrs==18.2.0
  Using cached https://files.pythonhosted.org/packages/3a/e1/5f9023cc983f1a628a8c2fd051ad19e76ff7b142a0faf329336f9a62a514/attrs-18.2.0-py2.py3-none-any.whl
Collecting colorama==0.4.0
  Using cached https://files.pythonhosted.org/packages/0a/93/6e8289231675d561d476d656c2ee3a868c1cca207e16c118d4503b25e2bf/colorama-0.4.0-py2.py3-none-any.whl
Collecting py==1.7.0
  Using cached https://files.pythonhosted.org/packages/3e/c7/3da685ef117d42ac8d71af525208759742dd235f8094221fdaafcd3dba8f/py-1.7.0-py2.py3-none-any.whl
ERROR: Could not find a version that satisfies the requirement pywin32==224 (from pycatia) (from versions: 225, 226, 227)
ERROR: No matching distribution found for pywin32==224 (from pycatia)
`

evereux, 2019-12-24:
I've updated pycatia now to `0.1.7` so Python 3.8+ users can pip install from pypi.

You may need to do `pip install pycatia --no-cache-dir` if pip is still trying to install `0.1.6`.

evereux, 2019-12-24:
Thanks for reporting this @wyfhust.

wyfhust, 2019-12-25:
Thank you for your excellent work.

closed 2019-12-25 by wyfhust