evereux/pycatia #12: How do i "New/CATPart"

GeorgeM987, 2020-01-23:
Hi,

Just started tinkering with this library but did not seem to find a way to get an CATPart/ CATProduct to open as New...is that possible currently? I'm using Catia V5R20 & Python 3.7 on Windows10. Thanks.

evereux, 2020-01-23:
Hi George,

You would use the `Documents.add('Part')` or `Documents.add('Product')`.

Here is a simple example:

    >>> from pycatia import CATIAApplication
    >>> catia = CATIAApplication()
    >>> documents = catia.documents()
    >>> documents.add('Part')

You should then see a new CATPart created in your CATIA session.

See https://pycatia.readthedocs.io/en/latest/api.html#pycatia.Documents.add although the formatting does seem to be messed up so this maybe easier to read: https://github.com/evereux/pycatia/blob/master/pycatia/document.py#L41.

GeorgeM987, 2020-01-23:
> Hi George,
>
> You would use the `Documents.add('Part')` or `Documents.add('Product')`.
>
> Here is a simple example:
>
> ```
> ```
>
> You should then see a new CATPart created in your CATIA session.
>
> See [...]

Thanks for that; i've done exactly that but nothing loaded in my Catia session for some reason and i got this errror:

"""**document_types = ['Part', 'Product', 'Drawing']
        if document_type not in document_types:
            raise ValueError(f'Document type must be in [{document_types}]')**"""

I am using VS Code if that makes any difference(doubt it though); can't really try anything on my work laptop as the Google's Collab. seems to be on Linux and Repl.it as well.

evereux, 2020-01-23:
Do any of the examples on the github repo main page work for you?

Can you show me the exact code you ran and full error log from console please?

GeorgeM987, 2020-01-23:
Yeah, i will do, once i'll get back home...and will try the other examples; as i was saying, just found about this last night and did not have enough time to go through the examples...just wanted to get a new part opened as a "proof of concept" kind of thing.
You're saying though it shouldn't be an issue with the R20 version of my Catia? i believed you've wrote somewhere it's been tested on R21.

evereux, 2020-01-23:
I managed to run the test suite successfully on everything from R21 up-to R27 I think it was last year at a company I was working at. There were no problems.

That error message you reported above suggests you didn't type 'Part', 'Product', or 'Drawing'. The case is important. Simply:

    >>> a = ['Part', 'Product', 'Drawing']
    >>> b = 'Part'
    >>> c = 'part'
    >>> b in a
    True
    >>> c in a
    False
    >>>

GeorgeM987, 2020-01-23:
I've done just what you've wrote:

from pycatia import CATIAApplication
catia = CATIAApplication()
documents = catia.documents()
documents.add('Part')

-probably just the variables have different naming...but exactly the same structure :) & that got me the **(f'Document type must be in [{document_types}]')** error in the console;
-you've mentioned R21 to R27(that's great as i'm using R26 at work-just need proper Python installed on my machine-) but have you tested R20?

evereux, 2020-01-23:
I don't think R20 was available but can't be sure as R18-R27 was available at one point but can't be sure I tested them. 

The only things that _should_ change between releases with regards to the CATIA COM / API interface are new features / bug fixes.

However your error is completely unrelated to the CATIA version. You'll have to forgive my scepticism but I reproduced your error by using a lowercase `part`.

    from pycatia import CATIAApplication
    catia = CATIAApplication()
    documents = catia.documents()
    documents.add('part')
    Traceback (most recent call last):
      File "<input>", line 4, in <module>
      File "C:\Users\paul\python-projects\pycatia\pycatia\document.py", line 73, in add
      raise ValueError(f'Document type must be in [{document_types}]')
    ValueError: Document type must be in [['Part', 'Product', 'Drawing']]

I think the next version of pycatia will be prevent the user from be able to make that mistake.

evereux, 2020-01-23:
Which version of pycatia are you using? Is it latest installed using pip or cloned from github?

GeorgeM987, 2020-01-23:
Yeah, i agree; i've pip installed.

GeorgeM987, 2020-01-23:
Right...so you were right all the way...seems my late night memory failed me; it was 'part' in lowercase... so i owe you an apology on that one; but still, now with no errors i still don't get the Part in my Catia session... any ideas?!

evereux, 2020-01-23:
Is CATIA already running? It kinda needs to be unless you launch CATIA in the script using whatever command line arguments you need to pick up your environment settings.

Other than that I've no idea. Especially if no com error is being thrown as that would suggest the CATIA COM interface wasn't happy with something. If that was the case I'd suggest reinstalling CATIA as I've seen one case where the CATIA installation on 1 machine out of around 15 in one office that didn't install the COM interface objects properly (think it was a registry problem).

GeorgeM987, 2020-01-23:
Yep, Catia already running in the background(tested a CATVbs macro and works fine); i suspected it might be the COM issue...as i've tested a piece of python code with pywin32:

import win32com.client.dynamic
CATIA = win32com.client.Dispatch("CATIA.Application")
catDocs = CATIA.Documents
newProductDoc = catDocs.Add("Part")

and had the same result...nothing appeared in my Catia session; i've read somewhere it could be possible to fix from within the bin folder -> cnext.exe /regserver but don't quite get how that's supposed to be done.

evereux, 2020-01-24:
If a macro works OK I don't think you need do anything. Maybe reinstall CATIA as a last resort?

Do the examples work for you?

GeorgeM987, 2020-01-24:
I haven't managed to run through the examples yet; plan to start looking a bit more into COM's and the way these work etc. & then will reinstall CATIA.

GeorgeM987, 2020-01-24:
So managed to do a quick re-install of my Catia and everything works now; thanks for all the support, happy to have this issue closed.

evereux, 2020-01-26:
Great! I'm glad you got it sorted and thanks for letting me know.

closed 2020-01-26 by evereux