evereux/pycatia #13: use pycatia to make sketch, surface, solid part, etc. features

luffyhan, 2020-02-14:
Hello, is it possible for pycatia to make sketch, surface, solid part, etc. features? Which include all functions in CAA V5? Thank you.

evereux, 2020-02-14:
Currently, no, (I am looking into this though).

If you are already familiar with scripting in CATIA using traditional methods you should then be able to look at the pycatia source code and figure out where to go from there. Otherwise, you'll have to wait for a future version.

Magic-Girl, 2020-05-23:
Hello, thank you very much for providing pycatia.
It solves a problem bothering me for a long time.
I probably understand the working principle of the library, but I'm not a professional programming background.
I really hope you can will continue to maintain this library.
If it can get to a subset of the "Geometrical set.", Perfect. Thanks again.

evereux, 2020-06-15:
> 
> 
> Hello, is it possible for pycatia to make sketch, surface, solid part, etc. features? Which include all functions in CAA V5? Thank you.

Take a look at example_013.py. It shows an example of some of the features you ask that are available in the latest version. I've not yet experimented with sketches ( maybe this week ).

I'm closing this since 'all functions of CAA V5' probably won't ever happen unless someone would like to finance that.

If there is a specific issue or problem you'd like to try and solve please raise another ticket.

closed 2020-06-15 by evereux

evereux, 2020-06-16:
example_016.py shows an example of a constrained sketch which is then padded.