evereux/pycatia #14: How to get the center of gravity coordinates of parts？

Zhigang-shrek, 2020-03-04:
I found that your test failed in your test file. Did you find any new methods？Appreciate

evereux, 2020-03-05:
Just cloned the repo and run the tests. That test does not fail here.

Zhigang-shrek, 2020-03-06:
You mean the last method in your test file?
Thank you!

evereux, 2020-03-06:
I think you're talking about this https://github.com/evereux/pycatia/blob/1cb3ddec309b7770f9214204c9d8982cb2fe5898/tests/test_measurable.py#L216

This isn't the last test and while I commented in the code that it failed all the tests do pass (this test is commented out). I wrote these tests sometime ago so I'd forgotten about it and your posts confused me.

I'll have a look at this and see if i can solve it.

Zhigang-shrek, 2020-03-06:
I'm really talking about this. I set part_body as reference but couldn't get the center.
I don't know whether it's my own problem or any other problem or not. After seeing the post, I found that you also encountered this problem, so I came to ask you.
Thanks again！

evereux, 2020-03-06:
This works for me.

    from pycatia import CATIADocHandler
    from pycatia import CATIAMeasurable
    from pycatia import create_measurable
    from pycatia import create_spa_workbench
    
    measurable_part = 'tests\\CF_catia_measurable_part.CATPart'
    
    
    
    with CATIADocHandler(measurable_part) as handler:
        catia = handler.catia
        document = handler.document
    
        center = (-47.039, 83.488, 0.0)
    
        spa_workbench = create_spa_workbench(document.document)
        part = document.part()
    
        body = part.get_body_by_name('PartBody')
    
        body_reference = part.create_reference(body)
        body_measurable = create_measurable(spa_workbench, body_reference)
    
        catia_measurable_body = CATIAMeasurable(body_measurable)
        catia_center = catia_measurable_body.get_cog(catia)
    
        print(catia_center)

Zhigang-shrek, 2020-03-07:
Does this method not apply in the products documentation?
For example, I want to get the center of gravity coordinates of a part in products.

Zhigang-shrek, 2020-03-07:
I want to get the center of gravity coordinates of each part in an assembly, and judge the relationship between each part through some algorithms. And I'm sorry that I didn't tell you that I used Visual Basic 6.0 to do CATIA development. Thank you for your reply. You don't need to help me. Let me find a way for myself. If I solve it, I'll let you know at the first time.

evereux, 2020-03-12:
I've just pushed some updates to the Master branch that'll help you do what I think you want. 

Take a look at example_10.py.

Zhigang-shrek, 2020-03-16:
I think I still need your help. This is my code written in VB:
`Dim part1`
`Set part1 = selection1.Item(1).Value`
`Dim ref1 As Reference`
`Set ref1 = part1.CreateReferenceFromObject(part1.MainBody)`
`Dim TheSPAWorkbench As Workbench`
`Set TheSPAWorkbench = CATIA.ActiveDocument.GetWorkbench("SPAWorkbench")`
`Dim TheMeasurable As Object`
`Set TheMeasurable = TheSPAWorkbench.GetMeasurable(ref1)`
`Dim VOL`
`VOL = TheMeasurable.Volume        'Return the volume of part1`
`Dim coords(2)`
`TheMeasurable.GetCOG (coords)`
`Dim x1, y1, z1`
`x1 = coords(0)`
`y1 = coords(1)`
`z1 = coords(2)`

Part1 is a part body, I can get the volume of part1 in this way. But I use GetCOG to get the center of gravity coordinates and return null values. I don't know where the problem is.

evereux, 2020-03-16:
Where is your python code? VB code is no help to me.

evereux, 2020-04-08:
I'm closing this as it doesn't seem to be related to pycatia.

closed 2020-04-08 by evereux