evereux/pycatia #18: how to load？

simonasdmd, 2020-04-07:
Hello

There are 3drep files in my 3DXML file.

How should this file be loaded.

[image]

evereux, 2020-04-07:
What problem with pycatia are you having exactly? What have you tried?

If you were opening these files in CATIA what would you do?

Can you provide examples files so that I may try and reproduce the problem?

simonasdmd, 2020-04-08:
hi~ drop-dead gorgeous'
I use CATIA to open the 3DXML file. Then save the file as well. Save to 3DS format.
I want to do this with pycatia.
Can pycatia be implemented?
With my test data.
[test.zip](https://github.com/evereux/pycatia/files/4447808/test.zip)

evereux, 2020-04-08:
That should work. Have you taken a look at the examples? There is one that shows you how to open a file and save it as another file type.

simonasdmd, 2020-04-09:
I tried it locally. Maybe it's because I'm not familiar. Still unsuccessful. Do you have examples of tests there? Can you give it a try.

evereux, 2020-04-09:
Example 8 shows how to open all .CATParts in a folder and save them as IGS. It should be quite straight forward to adapt that. However, when trying to do this manually with CATIA V5 I don't see an option to save as 3DS. Is that even possible? This is why I asked:

> If you were opening these files in CATIA V5 what would you do?

If you can't do it from within CATIA V5 itself, you can't do it with pycatia.

simonasdmd, 2020-04-09:
My software is this

[image]

[image]

simonasdmd, 2020-04-09:
Is this CATIA V5?

evereux, 2020-04-09:
No.

evereux, 2020-04-09:
Closing issue as one of the prerequisites of pycatia is not satisfied (having CATIA V5 actually installed). See https://github.com/evereux/pycatia/issues/19#issuecomment-611451098 for further information.

closed 2020-04-09 by evereux