evereux/pycatia #19: Installation failed

simonasdmd, 2020-04-08:
hi
I followed the installation instructions and an error occurred at the last step.
What is the cause.
[image]

evereux, 2020-04-08:
I'm assuming here you installed from pypi? It seems the 0.1.8 build is incomplete. Also, the docs need updating for that version. I'll get this fixed today.

Thanks for reporting.

evereux, 2020-04-08:
I've now updated pycatia.

Can you please try:
    pip install pycatia==0.2.0

Also, note the import command is now (docs are updataed, you may need to F5 and refresh the page/s).
    (env) C:\Users\<username>\python\pycatia-scripts>python
    Python 3.6.7 (v3.6.7:6ec5cf24b7, Oct 20 2018, 13:35:33) [MSC v.1900 64 bit (AMD64)] on win32
    Type "help", "copyright", "credits" or "license" for more information.
    >>> from pycatia.base_interfaces import CATIADocHandler    
    >>> help(CATIADocHandler)

simonasdmd, 2020-04-09:
I did solve yesterday's problem as you said. But as I continue to try "example" 2. Py. A new error has been encountered. It's my way of doing things, isn't it？

[image]

simonasdmd, 2020-04-09:
(env) C:\Users\zhangweiqiang\python\pycatia-scripts\env\Scripts>python
Python 3.8.2 (tags/v3.8.2:7b3ab59, Feb 25 2020, 22:45:29) [MSC v.1916 32 bit (Intel)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> from pycatia.base_interfaces import CATIADocHandler
>>> from pycatia.space_analyses_interfaces import CATIAMeasurable
>>> from pycatia.space_analyses_interfaces import create_measurable
>>> from pycatia.workbenches import create_spa_workbench
>>> catia = CATIAApplication()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name 'CATIAApplication' is not defined
>>> from pycatia.base_interfaces import CATIAApplication
>>> catia = CATIAApplication()
Traceback (most recent call last):
  File "C:\Users\zhangweiqiang\python\pycatia-scripts\env\lib\site-packages\win32com\client\dynamic.py", line 89, in _GetGoodDispatch
    IDispatch = pythoncom.connect(IDispatch)
pywintypes.com_error: (-2147221005, '无效的类字符串', None, None)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Users\zhangweiqiang\python\pycatia-scripts\env\lib\site-packages\pycatia\base_interfaces\catia_application.py", line 48, in __init__
    self.catia = Dispatch('CATIA.Application')
  File "C:\Users\zhangweiqiang\python\pycatia-scripts\env\lib\site-packages\win32com\client\__init__.py", line 95, in Dispatch
    dispatch, userName = dynamic._GetGoodDispatchAndUserName(dispatch,userName,clsctx)
  File "C:\Users\zhangweiqiang\python\pycatia-scripts\env\lib\site-packages\win32com\client\dynamic.py", line 114, in _GetGoodDispatchAndUserName
    return (_GetGoodDispatch(IDispatch, clsctx), userName)
  File "C:\Users\zhangweiqiang\python\pycatia-scripts\env\lib\site-packages\win32com\client\dynamic.py", line 91, in _GetGoodDispatch
    IDispatch = pythoncom.CoCreateInstance(IDispatch, None, clsctx, pythoncom.IID_IDispatch)
pywintypes.com_error: (-2147221005, '无效的类字符串', None, None)
>>>

evereux, 2020-04-09:
>  It's my way of doing things, isn't it？

Yes. :smile: 

Your code:

    from pycatia.base_interfaces import CATIADocHandler
Example 2 code:

    from pycatia.base_interfaces import CATIAApplication

evereux, 2020-04-09:
My bad! I didn't read through properly sorry.

Do you have CATIA V5 running? If so and this still happens my hunch is that the COM objects haven't been registered properly. Re-installation should solve this.

simonasdmd, 2020-04-09:
My software is this
[image]

[image]

simonasdmd, 2020-04-09:
Is this CATIA V5?

evereux, 2020-04-09:
> Is this CATIA V5?

No sir. That is CATIA Composer. First I've ever heard of it.

simonasdmd, 2020-04-09:
Can't I use pycatia?

simonasdmd, 2020-04-09:
I'm a lady. Very, very lovely lady..

evereux, 2020-04-09:
> Can't I use pycatia?

It would appear not ma'am. CATIA V5 is a per-requisite.

simonasdmd, 2020-04-09:
Is it OK for me to install a CATIA V5?
Is this version of CATIA v5r2019 OK？

simonasdmd, 2020-04-09:
Can the software you use open this data?
[JC.zip](https://github.com/evereux/pycatia/files/4455311/JC.zip)

simonasdmd, 2020-04-09:
I found out I had this
[image]

evereux, 2020-04-09:
I can't support anything other than issues with using pycatia. One of the requirements for that is having CATIA V5 installed. Will pycatia work with CATIA V5 r2019? I haven't and can't test that but it should. Will you be able to save a 3DXML to 3DS which you asked elsewhere? I don't think so but you have a later version of V5 available so maybe that can? I suspect your best bet is saving as CGR but we're veering off the topic for what this forum is for.

pycatia would typically be used to automate your CATIA V5 workflow. You must first find out if CATIA V5 itself can do what you expect before looking at pycatia. If you require support for CATIA V5 you can ask over at reddit: https://www.reddit.com/r/CATIA/.

I'm closing this topic as the installation issue itself has been resolved.

If you have further problems with pycatia (not CATIA V5 etc). Please open another thread but please first read the README file at: https://github.com/evereux/pycatia

closed 2020-04-09 by evereux

simonasdmd, 2020-04-09:
Oh, I see. Functions that CATIA V5 can achieve. Pycatia should be fine. If CATIA V5 does not have the ability itself, pycatia cannot.
Thank you very much. You've helped me a lot. I'll download V5 and try

evereux, 2020-04-09:
You're welcome and good luck. 

I browse the reddit CATIA community sometimes myself so if I can help with CATIA V5 there I will.

simonasdmd, 2020-04-09:
I think you can also try CATIA composer. Because it's not much different from CATIA V5.

simonasdmd, 2020-04-14:
@evereux 

Hello. I have installed CATIA V5 R20 now. Can I use pycatia for this version?

[image]

It can be "saved as" WRL format. Can we use pycatia to do this?

[image]

my test data:

[JC.zip](https://github.com/evereux/pycatia/files/4473947/JC.zip)

evereux, 2020-04-17:
> I have installed CATIA V5 R20 now. Can I use pycatia for this version?

That should be fine.

> It can be "saved as" WRL format. Can we use pycatia to do this?

That should work. Why not just try it?