evereux/pycatia #24: How to get to a subset of the "Geometrical set."？

Magic-Girl, 2020-05-23:
Hello, thank you very much for providing pycatia.
If it can get to a subset of the "Geometrical set.", Perfect. Thanks again

evereux, 2020-05-23:
What do you mean by subset? A hybrid body or hybrid shape?

Whatever it is you want it's best done with the development version as it's handled better there. It'll be a few weeks I think before it's released though. Take a look at this example: https://github.com/evereux/pycatia/blob/developement/example_13.py

You'll need to download the development version manually.

Magic-Girl, 2020-05-23:
Sorry, Sir.
My English is poor.
I don't know how to express my question to you.
So, I send you a picture, maybe you can understand it.
I am a Chinese student, I'm trying to learn Secondary Development of CATIA by Python.
Nice to meet you. Can you make friends?
Thanks.
[image]

Magic-Girl, 2020-05-23:
Sorry, Sir.
My English is poor.
I don't know how to express my question to you.
So, I send you a picture, maybe you can understand it.
I am a Chinese student, I'm trying to learn Secondary Development of CATIA by Python.
Nice to meet you. Can you make friends?
Thanks.

Please see the attachment.

evereux, 2020-05-24:
If you know the name upfront you could do something like this.

    from pycatia import catia
    
    
    def recursive_body_finder(hybrid_bodies, bodies_i_want, my_bodies):
        """
    
        Function to loop through the HybridBodies.items() list and find the construction set based on name.
    
        Args:
            hybrid_bodies: HybridBodies()
            bodies_i_want: ["bodie_names"]
            my_bodies: [HybridBody()]
    
        Returns:
    
        """
        for hybrid_body in hybrid_bodies:
            if hybrid_body.name in bodies_i_want:
                my_bodies.append(hybrid_bodies)
            # does the HybridBody() contain further HybridBodies() ?
            if hybrid_body.hybrid_bodies.count > 0:
                # yep!
                my_bodies = recursive_body_finder(hybrid_body.hybrid_bodies.items(), bodies_i_want, my_bodies)
    
        return my_bodies
    
    
    document = catia.active_document
    part = document.part()
    # Get the HybridBodies() object.
    hybrid_bodies = part.hybrid_bodies
    
    # get a list of hybrid_bodies.
    hybrid_bodies = hybrid_bodies.items()
    # print(hybrid_bodies)
    # [HybridBody(name="Geometrical Set.1"), HybridBody(name="Geometrical Set.2"), HybridBody(name="Geometrical Set.6")]
    
    # list of hybrid_body names we want.
    bodies_i_want = ["Geometrical Set.3", "Geometrical Set.4", "Geometrical Set.5"]
    # initialise the list.
    my_bodies = []
    my_bodies = recursive_body_finder(hybrid_bodies, bodies_i_want, my_bodies)
    
    print(my_bodies)

This is with the development version.

Magic-Girl, 2020-05-25:
Ok,thanks sir.By the way, your development version is probably what time completely released?

 

 from pycatia import catia def recursive_body_finder(hybrid_bodies, bodies_i_want, my_bodies):     """     Function to loop through the HybridBodies.items() list and find the construction set based on name.     Args:         hybrid_bodies: HybridBodies()         bodies_i_want: ["bodie_names"]         my_bodies: [HybridBody()]     Returns:     """     for hybrid_body in hybrid_bodies:         if hybrid_body.name in bodies_i_want:             my_bodies.append(hybrid_bodies)         # does the HybridBody() contain further HybridBodies() ?         if hybrid_body.hybrid_bodies.count &gt; 0:             # yep!             my_bodies = recursive_body_finder(hybrid_body.hybrid_bodies.items(), bodies_i_want, my_bodies)     return my_bodies document = catia.active_document part = document.part() # Get the HybridBodies() object. hybrid_bodies = part.hybrid_bodies # get a list of hybrid_bodies. hybrid_bodies = hybrid_bodies.items() # print(hybrid_bodies) # [HybridBody(name="Geometrical Set.1"), HybridBody(name="Geometrical Set.2"), HybridBody(name="Geometrical Set.6")] # list of hybrid_body names we want. bodies_i_want = ["Geometrical Set.3", "Geometrical Set.4", "Geometrical Set.5"] # initialise the list. my_bodies = [] my_bodies = recursive_body_finder(hybrid_bodies, bodies_i_want, my_bodies) print(my_bodies)

evereux, 2020-05-25:
:shrug: I couldn't say for sure. It's done when it's done. :smile:

evereux, 2020-06-15:
The version with this functionality has now been released.

closed 2020-06-15 by evereux