evereux/pycatia #25: Add type annotations to methods while scraping .chm help file

alwinw, 2020-07-02:
Hi, your package looks great! I had an enhancement suggestion which is to add [type hints](https://docs.python.org/3/library/typing.html) to the class methods generated which scraping the original .chm help files.

If you share your script used to scrape .chm files or add it to the repository (I couldn't find it, sorry), I'm happy to help out. It would be much easier than me trying to manually add them all in.

Thanks!

evereux, 2020-07-02:
I think the better solution is to scrape the existing project and pull the types from the doc_strings.

I'm very reluctant to share the scripts I used to scrape the chm files for a few reasons:
* They're a hack fest.
*  I don't want to document and support them.
* There are a few gotcha's in the results that I've fixed manually, several times, when merging back into pycatia. As a result, I don't plan at this stage on doing that again unless there is a really compelling reason to do so. It's tedious to say the least. :smile: 
* The chm file formatting is a mess (hence hack fest). How Dassault continues to release them like that I don't know. Something is broken their end. This is why I think it's better to do this using the pycatia .py files. It'll save some blood sweat and tears in the long run, trust me.

Personally, I'm not sure that adding type hints has many advantages for a project like this? That said, I'm only familiar with them from the limited reading I've done.

If you'd like to tackle this by all means do so. But please don't submit a pull request adding them to everything. Let's do this in small stages and tackle a few at a time. I recommend starting with `system_interfaces/any_object`, `system_interfaces/collection`, `in_interfaces/application` and `in_interfaces/document`.

Thanks for your interest and sharing your thoughts! :+1:

evereux, 2020-07-06:
If you take a look at the https://github.com/evereux/pycatia/blob/developement/v5automation-autogenerated.7z you'll see they've been updated to include type hints.

I updated the Application class `in_interfaces/application` and my tests still work so I guess I didn't break anything. :smiley:  I merged just the type hints and :rtypes: using pycharm's compare feature and it was a bit fiddly. Maybe there's a better, **safe**, way to do this automatically? Currently, pycharms merge isn't safe, at least for this class I tested. I'll think on it.

alwinw, 2020-07-08:
Looks great! Thanks for taking the time to make the changes. I've forked and started adding type hints, but I'll have a look through your `.7z` file. I wonder if `git merge` is safer than using pycharm, but yes I can imagine it being fidley. I'll put through a pull request with your other recommended files in the coming days 👍

evereux, 2020-07-08:
I've added type hints to some of the classes I mentioned above so make sure you update your branch.

> I wonder if git merge is safer than using pycharm, but yes I can imagine it being fidley. 

This is kind of what I ended up doing. I didn't realise you could commit chunks and this is much better. 

My workflow for this now is to 
* fork the developement branch. 
* Overwrite the .py files I want to update with type hints using the source from https://github.com/evereux/pycatia/blob/developement/v5automation-autogenerated.7z 
* commit the chunk with the type hint. (I don't want to commit any doc strings, they're a bit messy)
* stash the changes.
* merge my fork back into development.

It only becomes tricky when there are additional methods I've added. 

Let me know which folders you're not doing and I'll update those.

Cheers! :+1:

evereux, 2020-07-08:
ps if you have any CATVariant types you can `from pycatia.types import cat_variant` and use that for the CATVariant type. See `pycatia.product_structure_interfaces.publications` for an example. I think I've done it right anyway. :smile:

evereux, 2020-07-08:
So I've done the pycatia.drafting_interfaces directory now.

I'll not do any more for now as I want to work on the tests. I'd like the tests to generate the test files before running so anyone can run them regardless of the version of CATIA V5 they have.

evereux, 2020-07-11:
I'm now going to do the knowledge_interfaces directory.

@alwinw can you let me know the status of your changes please? ie which area's you are tackling? Thanks.

If pull the latest developement branch you should now be able to run the tests.

evereux, 2020-07-11:
I've done the knowledge_interfaces directory.

I'm now going to do the space_analyses_interfaces directory.

evereux, 2020-07-11:
I've done the space_analyses_interfaces directory.

I'll do the sketcher_interfaces next.

evereux, 2020-07-24:
This is all done now as far as I can tell so closing issue.

Thanks for planting the seed of the idea. 😄

closed 2020-07-24 by evereux

alwinw, 2020-07-25:
Hey, really sorry I didn't get back to you in time. I'm in Melbourne and things are pretty hectic cos of COVID - didn't have time to go through and test my changes in CATIA (don't have it on my home desktop). Thanks again for making the changes and for all your work!

evereux, 2020-07-25:
No worries and you're welcome. 👍 

The only reason I've been able to make all the significant updates to pycatia over the past few months is due to Covid.