evereux/pycatia #26: how to call a macro

shahram7, 2020-07-18:
Hello everyone

i have written a python function that writes a macro file which works correctly. 
Now I want to complete the function so that it calls the macro file too.
Could you please help me? Can pycatia do this?

`
def catcube(file_name, X, Y, Z):
    file_name = file_name
    x = "{:.6f}".format(X/2)
    y = "{:.6f}".format(Y/2)
    z = "{:.6f}".format(Z/2)

    import os
    import codecs
    import win32com.client
    import pycatia

    mcr = codecs.open(os.path.join(mcrlib, file_name+'.CATScript'), 'w', 'utf-8')
    mcr.write('Language="VBSCRIPT"\n\
\n\
Sub CATMain()\n\
\n\
Dim documents1 As Documents\n\
Set documents1 = CATIA.Documents\n\
\n\
Dim partDocument1 As Document\n\
Set partDocument1 = documents1.Add("Part")\n\
\n\
Dim part1 As Part\n\
Set part1 = partDocument1.Part\n\
\n\
Dim bodies1 As Bodies\n\
Set bodies1 = part1.Bodies\n\
\n\
Dim body1 As Body\n\
Set body1 = bodies1.Item("PartBody")\n\
\n\
Dim sketches1 As Sketches\n\
Set sketches1 = body1.Sketches\n\
\n\
Dim originElements1 As OriginElements\n\
Set originElements1 = part1.OriginElements\n\
\n\
Dim reference1 As AnyObject\n\
Set reference1 = originElements1.PlaneXY\n\
\n\
Dim sketch1 As Sketch\n\
Set sketch1 = sketches1.Add(reference1)\n\
\n\
Dim arrayOfVariantOfDouble1(8)\n\
arrayOfVariantOfDouble1(0) = 0.000000\n\
arrayOfVariantOfDouble1(1) = 0.000000\n\
arrayOfVariantOfDouble1(2) = 0.000000\n\
arrayOfVariantOfDouble1(3) = 1.000000\n\
arrayOfVariantOfDouble1(4) = 0.000000\n\
arrayOfVariantOfDouble1(5) = 0.000000\n\
arrayOfVariantOfDouble1(6) = 0.000000\n\
arrayOfVariantOfDouble1(7) = 1.000000\n\
arrayOfVariantOfDouble1(8) = 0.000000\n\
sketch1.SetAbsoluteAxisData arrayOfVariantOfDouble1\n\
\n\
part1.InWorkObject = sketch1\n\
\n\
Dim factory2D1 As Factory2D\n\
Set factory2D1 = sketch1.OpenEdition()\n\
\n\
Dim geometricElements1 As GeometricElements\n\
Set geometricElements1 = sketch1.GeometricElements\n\
\n\
Dim axis2D1 As GeometricElement\n\
Set axis2D1 = geometricElements1.Item("AbsoluteAxis")\n\
\n\
Dim line2D1 As CATBaseDispatch\n\
Set line2D1 = axis2D1.GetItem("HDirection")\n\
\n\
line2D1.ReportName = 1\n\
\n\
Dim line2D2 As CATBaseDispatch\n\
Set line2D2 = axis2D1.GetItem("VDirection")\n\
\n\
line2D2.ReportName = 2\n\
\n\
Dim point2D1 As Point2D\n\
Set point2D1 = factory2D1.CreatePoint(-' + x +', ' + y +')\n\
\n\
point2D1.ReportName = 3\n\
\n\
Dim point2D2 As Point2D\n\
Set point2D2 = factory2D1.CreatePoint(' + x +', ' + y +')\n\
\n\
point2D2.ReportName = 4\n\
\n\
Dim line2D3 As Line2D\n\
Set line2D3 = factory2D1.CreateLine(-' + x +', ' + y +', ' + x +', ' + y +')\n\
\n\
line2D3.ReportName = 5\n\
\n\
line2D3.StartPoint = point2D1\n\
\n\
line2D3.EndPoint = point2D2\n\
\n\
Dim point2D3 As Point2D\n\
Set point2D3 = factory2D1.CreatePoint(' + x +', -' + y +')\n\
\n\
point2D3.ReportName = 6\n\
\n\
Dim line2D4 As Line2D\n\
Set line2D4 = factory2D1.CreateLine(' + x +', ' + y +', ' + x +', -' + y +')\n\
\n\
line2D4.ReportName = 7\n\
\n\
line2D4.EndPoint = point2D2\n\
\n\
line2D4.StartPoint = point2D3\n\
\n\
Dim point2D4 As Point2D\n\
Set point2D4 = factory2D1.CreatePoint(-' + x +', -' + y +')\n\
\n\
point2D4.ReportName = 8\n\
\n\
Dim line2D5 As Line2D\n\
Set line2D5 = factory2D1.CreateLine(' + x +', -' + y +', -' + x +', -' + y +')\n\
\n\
line2D5.ReportName = 9\n\
\n\
line2D5.StartPoint = point2D3\n\
\n\
line2D5.EndPoint = point2D4\n\
\n\
Dim line2D6 As Line2D\n\
Set line2D6 = factory2D1.CreateLine(-' + x +', -' + y +', -' + x +', ' + y +')\n\
\n\
line2D6.ReportName = 10\n\
\n\
line2D6.EndPoint = point2D4\n\
\n\
line2D6.StartPoint = point2D1\n\
\n\
Dim constraints1 As Constraints\n\
Set constraints1 = sketch1.Constraints\n\
\n\
Dim reference2 As Reference\n\
Set reference2 = part1.CreateReferenceFromObject(line2D3)\n\
\n\
Dim reference3 As Reference\n\
Set reference3 = part1.CreateReferenceFromObject(line2D1)\n\
\n\
Dim constraint1 As Constraint\n\
Set constraint1 = constraints1.AddBiEltCst(catCstTypeHorizontality, reference2, reference3)\n\
\n\
constraint1.Mode = catCstModeDrivingDimension\n\
\n\
Dim reference4 As Reference\n\
Set reference4 = part1.CreateReferenceFromObject(line2D5)\n\
\n\
Dim reference5 As Reference\n\
Set reference5 = part1.CreateReferenceFromObject(line2D1)\n\
\n\
Dim constraint2 As Constraint\n\
Set constraint2 = constraints1.AddBiEltCst(catCstTypeHorizontality, reference4, reference5)\n\
\n\
constraint2.Mode = catCstModeDrivingDimension\n\
\n\
Dim reference6 As Reference\n\
Set reference6 = part1.CreateReferenceFromObject(line2D4)\n\
\n\
Dim reference7 As Reference\n\
Set reference7 = part1.CreateReferenceFromObject(line2D2)\n\
\n\
Dim constraint3 As Constraint\n\
Set constraint3 = constraints1.AddBiEltCst(catCstTypeVerticality, reference6, reference7)\n\
\n\
constraint3.Mode = catCstModeDrivingDimension\n\
\n\
Dim reference8 As Reference\n\
Set reference8 = part1.CreateReferenceFromObject(line2D6)\n\
\n\
Dim reference9 As Reference\n\
Set reference9 = part1.CreateReferenceFromObject(line2D2)\n\
\n\
Dim constraint4 As Constraint\n\
Set constraint4 = constraints1.AddBiEltCst(catCstTypeVerticality, reference8, reference9)\n\
\n\
constraint4.Mode = catCstModeDrivingDimension\n\
\n\
sketch1.CloseEdition \n\
\n\
part1.InWorkObject = sketch1\n\
\n\
part1.Update \n\
\n\
Dim shapeFactory1 As Factory\n\
Set shapeFactory1 = part1.ShapeFactory\n\
\n\
Dim pad1 As Pad\n\
Set pad1 = shapeFactory1.AddNewPad(sketch1, 70.000000)\n\
\n\
Dim limit1 As Limit\n\
Set limit1 = pad1.FirstLimit\n\
\n\
Dim length1 As Length\n\
Set length1 = limit1.Dimension\n\
\n\
length1.Value = '+ z +'\n\
\n\
pad1.IsSymmetric = True\n\
\n\
part1.Update \n\
\n\
Set partDocument1 = CATIA.ActiveDocument\n\
\n\
partDocument1.SaveAs')
    mcr.write(' "'+os.path.join(os.getcwd(),file_name)+'.CATPart"\n\
\n\
End Sub\n')
    mcr.close()
`
Now my macro is ready.
I tried the following lines to call this macro but no success.

`
    #option1
    catia = win32com.client.Dispatch('CATIA.Application')
    catia.StartCommand(file_name + '.CATScript')
`
`
    #option2
    os.chdir(r'C:\Program Files\Dassault Systemes\B24\win_b64\code\bin')
    os.system('cnext CATIA.V5R16.B16 -macro'+ os.path.join(os.getcwd(),file_name+'.CATScript'))
`
Now I would ask if it is possible to call the macro file with pycatia.
Regards

evereux, 2020-07-18:
Yes you can do this. Take a look at the methods execute_script or evaluate. If you use evaluate you won't have to write the CATScript file.

    from pycatia import catia
    from pycatia.enumeration.enumeration_types import cat_script_language
    
    message_box_text = 'pycatia is great'
    vba_function_name = 'message_box'
    vba_code = """
    Public Function message_box(message_box_text)
            MsgBox(message_box_text)
    End Function
    """
    
    system_service = catia.system_service
    system_service.evaluate(vba_code, cat_script_language.index("CATVBScriptLanguage"), vba_function_name,
                        [message_box_text])

If you're going to use pycatia to do this you may as well just rewrite your script to just use pycatia and not call an external script.

shahram7, 2020-07-18:
thank you very much.
i am a beginner and do not know how to rewrite the macro to python. So I have decided to go this way.
Is there any tutorial for it?

evereux, 2020-07-18:
Take a look at the examples / documentation.

closed 2020-07-18 by shahram7