evereux/pycatia #27: How to get the user defined properties in part or product?

evereux, 2020-07-19:
How to get the user defined properties in part or product?

i am  trying with "user_ref_properties", its throwing me a com  error like below.

com_error: (-2147352567, 'Exception occurred.', (0, 'CATIAProduct', 'The method UserRefProperties failed', None, 0, -2147467259), None)

Can any one help me in that?

_Originally posted by @tcrakshi in https://github.com/evereux/pycatia/issues/10#issuecomment-660596608_

evereux, 2020-07-19:
@tcrakshi

I can't reproduce your error with the information you have provided. The method works fine here.

	from pycatia import catia

	documents = catia.documents
	document = catia.active_document
	product = document.product()
	user_properties = product.user_ref_properties

	user_property_1 = user_properties.item(1)

	print(user_property_1.name)
	print(user_property_1.value)

This outputs

    Part1\Properties\user_prop
    test

Please provide the code you run.

tcrakshi, 2020-07-19:
hi,
Thanks for quick reply,
my requirement is to fetch the user defined properties of all catpart/ assembly. 
Assembly contain n number of sub assembly, i need to run through all level and fetch  the user defined properties of part and  assembly.

my Code pasted below:

**from pycatia import catia
from pycatia.enumeration.enumeration_types import cat_work_mode_type
catia.display_file_alerts = False
########## Part level Extract
def getallnode(iProduct):
    catia.active_document.selection.clear()
    iProduct.apply_work_mode(cat_work_mode_type.index("DESIGN_MODE"))
    products = iProduct.get_products()
    for item in products:
        if item.is_catpart():
            user_properties  = item.user_ref_properties
        if item.is_catproduct():
            user_properties  = item.user_ref_properties
            getallnode(item)
           
##########    Main Function
rootProduct = 0
try:
    documents = catia.active_document
    rootProduct = documents.product()

except:
    print ("Product is not open")
    sys.exit()
else:
    print("Product Documnet found")
    
getallnode(rootProduct)**

Error:
com_error: (-2147352567, 'Exception occurred.', (0, 'CATIAProduct', 'The method UserRefProperties failed', None, 0, -2147467259), None)

tcrakshi, 2020-07-19:
thanks for your quick reply, below is the my code.
1.  i want sub assembly user defined properties, and sub level is not
defined, it will vary always.

*from pycatia import catiafrom pycatia.enumeration.enumeration_types import
cat_work_mode_typecatia.display_file_alerts = False*

*########## Part level Extractdef getallnode(iProduct):
catia.active_document.selection.clear()
iProduct.apply_work_mode(cat_work_mode_type.index("DESIGN_MODE"))
products = iProduct.get_products()    for item in products:        if
item.is_catpart():            user_properties = item.user_ref_properties
        print(user_properties.count)            ##########    Main
FunctionrootProduct = 0try:    documents = catia.active_document
rootProduct = documents.product()#    prop =
rootProduct.user_ref_properties#    first =
prop.get_item_by_name("78566533\Properties\CATIA_NAME")#    print("")#
 print(first.name <http://first.name>)#    print(first.value)except:
print ("Product is not open")    sys.exit()else:    print("Product Documnet
found")*

getallnode(rootProduct)

Regards
Rakshith Gowda

evereux, 2020-07-19:
Can you try reformatting your code please?

github uses [markdown](https://guides.github.com/features/mastering-markdown/) so when pasting code you need to indent by 4 spaces.

So, what I do is copy code, paste into text editor, select it all, then press tab.

Also, I can't see where your code is failing as you haven't provided the full trace back.

tcrakshi, 2020-07-19:
Regards
Rakshith Gowda

tcrakshi, 2020-07-19:
Hi,
i want change the position of product, i have all the 12 component.
Can any one provide me some  snippet of "set_components"  method.

evereux, 2020-07-19:
> 
> 
> Hi,
> i want change the position of product, i have all the 12 component.
> Can any one provide me some snippet of "set_components" method.

Please create a new issue for this.

evereux, 2020-07-19:
Can I recommend you go through some python tutorials and learn how to write more pythonic code. :-)

Please, also look at how the code is formatted below compared to what you provided.

Anyway, the problem was that you need to get the reference product of the product to access it's user_ref_properties. I might not sure if there is a quirk of pycatia or something baked into CATIA. I'll think about this.

	from pycatia import catia
	from pycatia.enumeration.enumeration_types import cat_work_mode_type

	def loop_products(cat_product):
		products = cat_product.get_products()

		for product_child in products:
			reference_product = product_child.reference_product

			if product_child.is_catpart():
				user_properties = reference_product.user_ref_properties
				print(user_properties)
			if product_child.is_catproduct():
				user_properties = reference_product.user_ref_properties
				print(user_properties)
				# this is a product so loop through again.
				loop_products(product_child)

	if __name__ == "__main__":
		document = catia.active_document
		product = document.product()

		# don't apply this at the function level, it only needs to be done once.
		product.apply_work_mode(cat_work_mode_type.index("DESIGN_MODE"))

		loop_products(product)

evereux, 2020-07-22:
No feedback and have provided working example so closing issue.

closed 2020-07-22 by evereux

tcrakshi, 2020-07-22:
thanks , now i am able to fetch user defined properties, thanks for your support

ptm-tm, 2024-10-03:
> @tcrakshi
>
> I can't reproduce your error with the information you have provided. The method works fine here.
>
> ```
> from pycatia import catia
>
> documents = catia.documents
> document = [...]

must be `product = document.product` for part

evereux, 2024-10-03:
> must be `product = document.product` for part

Whilst this is true now it wasn't when this issue was raised, 4 years ago. The issue has long since been closed and the author deleted the original issue anyway.

ptm-tm, 2024-10-03:
> > must be `product = document.product` for part
> 
> Whilst this is true now it wasn't when this issue was raised, 4 years ago. The issue has long since been closed and the author deleted the original issue anyway.

its not matter for me.