evereux/pycatia #28: Raising Issues / Asking Questions.

evereux, 2020-07-19:
Please don't raise an issue here until you have read the documentation, understood and met the requirements (YOU NEED CATIA INSTALLED) and have run the examples.

If you are upgrading from an earlier versions and your scripts break please check the changelog to give you clues on what to do to resolve those issues.

Once you have done that please give as much information as you can so I may try and reproduce the issue. Please provide the following information:

* Windows OS version
* CATIA version
* Python version
* pycatia version
* Your code!
* Full stack trace of error you encounter (properly formatted text please, not picture).

Thanks! :-)

evereux, 2020-07-24:
github issues are normally a place to report issues with a script that isn't
working as expected or issues with pycatia itself. Whilst I'm happy to support
new users to get them up and running I can't continue to provide technical 
support for questions that ask "Can pycatia do x?". Or simply "How do I?".
pycatia is currently, free open source software that can potentially save
companies many man hours and does not come with technical support.

If it's documented in the CAA.chm file there is a good chance it is or could 
be. Find what the class name of the method you want to use and search the 
`API <https://pycatia.readthedocs.io/en/latest/api_index.html>`_ for it. Or, do
the same with the source code in your IDE.

Using the class methods in pycatia will be very similar to how one would do so
in CATScript / VBScript.

If your script isn't working please post that script here, properly formatted,
so I can run it and we can find where the problem lies. Please strip the script
down to the bare essentials before you do so.

If you would like technical support for pycatia please contact me 
evereux@gmail.com, from a business address, and we can discuss contracting my
technical support for your company.

I'm closing this ticket. If some code is provided please feel free to re-open.

xuweilin-hub, 2021-01-19:
I can't find the API addnewtranslate2 in pycatia. It seems that the API is missing in the help document of CATIA. If you can, please supplement it. Thank you!

Rocasar, 2021-09-08:
Hi, thank you for your work with pycatia. It is amazing and really usefull. I would love to know if there is an implementation of the Multi-Section surface/Solid tool. I've been searching and I haven't found anything. 
Thank you !!

onixenic, 2021-09-26:
Is this class usable?

class pycatia.knowledge_interfaces.units_sheet_setting_att.UnitsSheetSettingAtt(com_object)

Which com_object am I supposed to use?

wheels1991, 2021-11-07:
I found a bug as following:catia().documents.item("string") return type should be partDocument not document

wheels1991, 2021-11-07:
I get a Document by Document.item(cat_variant), and then, how can I convert  Document to PartDocument or ProductDocument?