evereux/pycatia #29: Change product position

tcrakshi, 2020-07-19:
Hi,
i want change the position of product, i have all the 12 component.
Can any one provide me some snippet of "set_components" method.

evereux, 2020-07-20:
> i want change the position of product, 

The tests show an example of moving a product. https://github.com/evereux/pycatia/blob/afa21f4213a3507db01c365ea950e224accd7262/tests/test_product.py#L156

> i have all the 12 component.

? 

> Can any one provide me some snippet of "set_components" method.

It would probably be something like this:

    from pycatia.in_interfaces.position import Position
    
    # create move object, see link above
    # ....
    #
    
    position = Position(move)
    transformation = (
        1.000, 0, 0,
        0, 0.707, 0.707,
        0, -0.707, 0.707,
        10.000, 20.000, 30.000
    )
    position.set_components(transformation)

tcrakshi, 2020-07-21:
when i use set_components its giving me a error like "TypeError: set_components() takes 1 positional argument but 2 were given:
please do a needful help

My Code Pasted Below:

from pycatia import catia
from pycatia.in_interfaces.position import Position
from pycatia.enumeration.enumeration_types import cat_constraint_type, cat_constraint_mode, cat_constraint_angle_sector, cat_work_mode_type

     
def loop_product(iproduct):
    products  = iproduct.get_products()
    for item in products:
        pos = Position(item)
        component = (1 , 0 , 0 , 
                     0 , 1 , 0 ,
                     0 , 0 , 1 ,
                     0 , 0 , 0
                     )
        pos.set_components(component)

#### Main Function
document = catia.active_document
product = document.product()
product.apply_work_mode(cat_work_mode_type.index("DESIGN_MODE"))
loop_product(product)

Error:
runfile('C:/Users/A337885/Desktop/PY/untitled3.py', wdir='C:/Users/A337885/Desktop/PY')
Traceback (most recent call last):

  File "<ipython-input-7-cbe849f76d4a>", line 1, in <module>
    runfile('C:/Users/A337885/Desktop/PY/untitled3.py', wdir='C:/Users/A337885/Desktop/PY')

  File "C:\Anaconda3\lib\site-packages\spyder_kernels\customize\spydercustomize.py", line 827, in runfile
    execfile(filename, namespace)

  File "C:\Anaconda3\lib\site-packages\spyder_kernels\customize\spydercustomize.py", line 110, in execfile
    exec(compile(f.read(), filename, 'exec'), namespace)

  File "C:/Users/A337885/Desktop/PY/untitled3.py", line 22, in <module>
    loop_product(product)

  File "C:/Users/A337885/Desktop/PY/untitled3.py", line 16, in loop_product
    pos.set_components(component)

TypeError: set_components() takes 1 positional argument but 2 were given

evereux, 2020-07-21:
I was wrong above. This works for me.

    from pycatia import catia
    
    document = catia.active_document
    top_product = document.product()
    
    products = top_product.products
    first_product = products.item(1)
    print(first_product)
    
    new_position = (1, 0, 0,
                    0, 1, 0,
                    0, 0, 1,
                    0, 0, 0
                    )
    
    first_product.position.set_components(new_position)

Please learn how to format code by reading https://guides.github.com/features/mastering-markdown/.

tcrakshi, 2020-07-21:
i tried same code above mentioned:
still i am getting error like: return Product(self.products.Item(i_index.com_object))
AttributeError: 'int' object has no attribute 'com_object'
i am using CATIA R27 , Spyder(Python 3.7 ) , Windows 10  enterprisse ,64 bit.

i dont what is wrong in code, could you help me in that.

Traceback:

runfile('C:/Users/A337885/Desktop/PY/b.py', wdir='C:/Users/A337885/Desktop/PY')
Traceback (most recent call last):

  File "<ipython-input-9-daae4eb3b924>", line 1, in <module>
    runfile('C:/Users/A337885/Desktop/PY/b.py', wdir='C:/Users/A337885/Desktop/PY')

  File "C:\Anaconda3\lib\site-packages\spyder_kernels\customize\spydercustomize.py", line 827, in runfile
    execfile(filename, namespace)

  File "C:\Anaconda3\lib\site-packages\spyder_kernels\customize\spydercustomize.py", line 110, in execfile
    exec(compile(f.read(), filename, 'exec'), namespace)

  File "C:/Users/A337885/Desktop/PY/b.py", line 6, in <module>
    first_product = products.item(3)

  File "C:\Anaconda3\lib\site-packages\pycatia\product_structure_interfaces\products.py", line 247, in item
    return Product(self.products.Item(i_index.com_object))

AttributeError: 'int' object has no attribute 'com_object'

evereux, 2020-07-21:
Looks like you are running an older version of pycatia.
```
python -m pip install pycatia --upgrade
```

evereux, 2020-07-21:
Great. Glad you got it working! :smile:

closed 2020-07-21 by evereux