evereux/pycatia #32: How to call  the python code in catia?

tcrakshi, 2020-07-21:
How to call  the python code in catia?
if catia user dont have python editor in his machine, how it will work?
is there any mechanism to tackles this issue?

because native catia user dont have background of coding , so how we can deploy too the user to launch the python code in Catia interface. (Like catia native commands)

Regards,
Rakshith

evereux, 2020-07-21:
You can use py2exe (or similar) to compile an executable for the end user to run if they don't have python. Be warned though some companies virus checkers don't like the compiled result.

Otherwise, you have to have python installed.

evereux, 2020-07-21:
ps I can't support how you would do that right now.

tcrakshi, 2020-07-21:
ya, i am also thinking to do in pyinstaller or py2exe .

evereux, 2020-07-21:
Hmmm, this has got me thinking. It might be a good idea to provide a windows binary as a part of this project ...

evereux, 2020-07-24:
pycatia now has a win32 binary in the repository. See https://pycatia.readthedocs.io/en/latest/windows_builds.html.

evereux, 2020-07-24:
I'm closing this ticket. The request has been satisfied.

If there are issues with the executable please open a new ticket.

closed 2020-07-24 by evereux