evereux/pycatia #33: Generate CATPart From Product

tcrakshi, 2020-07-21:
is it possible to do Generate CATPart From Product in python code?
if it is possible, can anyone share some snippet of that code?

evereux, 2020-07-21:
It is not currently possible with pycatia but you could hack it. It seems that whilst the CATIA COM interface can do it it's not documented thus wasn't imported.

reference:  http://www.coe.org/p/fo/et/thread=28139

I may add this at a later date.

tcrakshi, 2020-07-21:
thanks for reply, please keep me inform, once you added this feature in catia. it is very useful.

evereux, 2020-07-21:
Agreed. I've had to create ALLCATParts many many times over the years.

I'll leave this ticket open and report progress in the future. Time permitting I may tackle this this week

evereux, 2020-07-24:
This has been added to pycatia 0.3.9. See https://pycatia.readthedocs.io/en/latest/api/pycatia/product_structure_interfaces/product.html#pycatia.product_structure_interfaces.product.Product.generate_ALLCATPart

usage would be ``product.generate_ALLCATPart(product)`` which would return a document object.

tcrakshi, 2020-07-24:
hey, its very good to see that command in pycatia.
i will use and give feedback soon

tcrakshi, 2020-07-26:
May i know, when pycatia 0.3.9 will be release?

evereux, 2020-07-26:
I forgot to upload to pypi! Sorry about that. I've done that now.

tcrakshi, 2020-08-04:
hi

i am trying to generate_ALLCATPart , but no luck.
i am using pycatia 0.4.0 version,  please do a needful help.

Requirement: Generate the catprt for root product or main product

**my Code:**
```Python code
from pycatia import catia

documents = catia.active_document
rootProduct = documents.product()
generateCatPart = rootProduct.generate_ALLCATPart(rootProduct)

```
**Error:**
```
Traceback (most recent call last):

  File "<ipython-input-4-9beb400b0747>", line 1, in <module>
    runfile('C:/Users/A337885/Desktop/PY/untitled2.py', wdir='C:/Users/A337885/Desktop/PY')

  File "C:\Anaconda3\lib\site-packages\spyder_kernels\customize\spydercustomize.py", line 827, in runfile
    execfile(filename, namespace)

  File "C:\Anaconda3\lib\site-packages\spyder_kernels\customize\spydercustomize.py", line 110, in execfile
    exec(compile(f.read(), filename, 'exec'), namespace)

  File "C:/Users/A337885/Desktop/PY/untitled2.py", line 5, in <module>
    generateCatPart = rootProduct.generate_ALLCATPart(rootProduct)

AttributeError: 'Product' object has no attribute 'generate_ALLCATPart'

```

low-life-products, 2020-08-04:
See the changelog for version 0.4.

tcrakshi, 2020-08-04:
i gone through  changelog for version 0.4 and .39 still not working.
can you guide me where i am doing mistake or if provide some snippet, it would be very useful.

**My code:**
``` python code
from pycatia import catia
caa = catia()
document = caa.active_document
rootProduct = document.product()
part = rootProduct.generate_ALLCATPart(rootProduct)

``` 
**error:**
``` 
runfile('C:/Users/A337885/Desktop/PY/untitled3.py', wdir='C:/Users/A337885/Desktop/PY')
Traceback (most recent call last):

  File "<ipython-input-17-cbe849f76d4a>", line 1, in <module>
    runfile('C:/Users/A337885/Desktop/PY/untitled3.py', wdir='C:/Users/A337885/Desktop/PY')

  File "C:\Anaconda3\lib\site-packages\spyder_kernels\customize\spydercustomize.py", line 827, in runfile
    execfile(filename, namespace)

  File "C:\Anaconda3\lib\site-packages\spyder_kernels\customize\spydercustomize.py", line 110, in execfile
    exec(compile(f.read(), filename, 'exec'), namespace)

  File "C:/Users/A337885/Desktop/PY/untitled3.py", line 24, in <module>
    part = rootProduct.generate_ALLCATPart(rootProduct)

  File "C:\Users\A337885\AppData\Roaming\Python\Python37\site-packages\pycatia\product_structure_interfaces\product.py", line 946, in generate_ALLCATPart
    part = product.com_object.GetItem("DECProductToPart")

  File "<COMObject <unknown>>", line 2, in GetItem

com_error: (-2147352567, 'Exception occurred.', (0, 'CATIAProduct', 'The method GetItem failed', None, 0, -2147467263), None)

```

evereux, 2020-08-04:
Can you manually try activating terminal node and activating design mode on the root product first and then run the script again please? I can't test myself just now.

If this is the issue I'll make pycatia throw a nicer error / hint to the user in the next version.

tcrakshi, 2020-08-04:
before running .py script, i have done design mode and activate terminal node as well. still not working

evereux, 2020-08-04:
OK.I'll test this out in the morning.

evereux, 2020-08-05:
I created a new product with one part in it that contained a solid block.

Your script ran fine.

If I run the script on a Part I get your error.

```python

from pycatia import catia

caa = catia()
document = caa.active_document
if document.is_product:
    product = document.product()
    # generate_ALLCATPart() returns a Document object 
    # so best name it document so it makes more sense in the future.
    document = product.generate_ALLCATPart(product)
else:
    print(f"Document '{document.name}' is not a product!")
```

tcrakshi, 2020-08-05:
I  ranned same script above what you have provided, still getting same error.
i tried in exciting and new  product still no luck 

i am using R27 catia,  windows10 , python 3.7.

if it is possible, can we have a call through skype or webex?

regrds,
rakshith

tcrakshi, 2020-08-05:
i am getting error in this line: 
**part = product.com_object.GetItem("DECProductToPart")**  in product.py file

``` python 
@staticmethod
    def generate_ALLCATPart(product: 'Product') -> 'Document':
        """

        Generate an ALLCATPart (CATPart) from CATProduct.

        :param Product product:
        :return: Document
        :rtype: Document
        """

        from pycatia.in_interfaces.document import Document

        part = product.com_object.GetItem("DECProductToPart")
        part.Run()
        part = part.GetResult()

        return Document(part)
```

low-life-products, 2020-08-05:
Is think this is possibly a CATIA V5 license issue.

tcrakshi, 2020-08-05:
what licence required?

evereux, 2020-08-05:
This is the one I believe: https://www.3ds.com/products-services/catia/products/v5/portfolio/domain/Infrastructure/product/DF1/?woc=%7B%22category%22%3A%5B%22category%2Fchampions%22%5D%7D&wockw=card_content_cta_1_url%3A%22https%3A%2F%2Fblogs.3ds.com%2Fcatia%2F%22

tcrakshi, 2020-08-05:
Hi,
yes you are correct,  its required DF1 lic.  
i checked with DF1 lic, now its working.

just thought:
if pycatia  generate all catpart works without DF1 licence, it will be more useful.

regards,
Rakshith

evereux, 2020-08-05:
Hmmm. I guess this would technically be possible ... loop through each
product > part, copy all visible elements, paste into target part.

Why not write a script yourself and share it? :-) I'll add it to the
examples and if it proves robust I'll add it to pycatia.

The current method will always use the license DF1. I'll add a check
for the license in the next version so that the user gets a warning.

tcrakshi, 2020-08-05:
yes, that would be great:

i already have a code in VB script, but there cons are more
1. if data size is more it will take time
2. position : if we go for copying on individual visual part level, there will be problem in instance ( need to take care and need to build logic)

now i am slowly getting into python, yes in future i will give try in python.

tcrakshi, 2020-08-12:
hi, 
i have question regarding generate cat product
1. is it will work for only root product or is it capable of work on selected product?

tcrakshi, 2020-08-12:
ya, i got it, it will work for selected product also

evereux, 2020-09-03:
I'm closing this now as I believe the original question has been answered.

closed 2020-09-03 by evereux

VBG8, 2022-05-17:
Tcrakshi, how did you manage to transform a product within a rootproduct into a catpart?