evereux/pycatia #34: Is there any way to return the endpoints of a point to point line?

andrewellis55, 2020-07-21:
The points and lines coulee be defined in CATIA like so.
[image]

evereux, 2020-07-21:
Yes there is. Take a look at the examples, particularly with regards to using the measurable object.

andrewellis55, 2020-07-21:
Thanks for the reply! I should have been a little more clear. I've used the measurable object that you mentioned, for example something like this.

```
from pycatia import catia

documents = catia.documents
documents.open(r'Part3_noname.CATPart')
document = catia.active_document
spa_workbench = document.spa_workbench()
part = document.part()

selected = document.search_for_items(['Line'])

for selection in selected:
    reference = part.create_reference_from_object(selection)
    measurable = spa_workbench.get_measurable(reference)

    points = measurable.get_points_on_curve()
    print(points)
```

This returns the locations of the endpoints and midpoint (which is very useful and I can definitely work around from those). But is it possible to return the item reference of the points themselves? If not its not an issue to just loop though the points to find which ones match up their coordinates to the outputs.

evereux, 2020-07-21:
Thanks for the clarification. :smile: 

Good question! I'm going to have to think about that. My gut instinct is currently no if it's an already created object, this is something that would require a update to pycatia.

I can think of an intermediate hack though. I'll get back to you tomorrow.

evereux, 2020-07-22:
```python

from pycatia import catia
from pycatia.hybrid_shape_interfaces.hybrid_shape_line_pt_pt import HybridShapeLinePtPt
from pycatia.hybrid_shape_interfaces.point import Point

document = catia.active_document

part = document.part()
hbs = part.hybrid_bodies

# get the hybrid body
hb_construction = hbs.item("ConstructionGeometry")

# get all the shapes
shapes = hb_construction.hybrid_shapes

# find the line by name
hs_line = shapes.item("Line.1")

# create a HybridShapeLinePtPt object from item reference.
hs_line_pt_pt = HybridShapeLinePtPt(hs_line.com_object)

# get the origin_points display name.
origin_point_name = hs_line_pt_pt.pt_origin.display_name

# get the point in the 'shapes' hybrid_shapes object using the reference points
# display_name and create a generic Point object from it.
hs_point_origin = Point(shapes.item(origin_point_name).com_object)
print(hs_point_origin.get_coordinates())
```

Take a look at this. I think this will point you in the right direction.

It's definitely hacky but there's nothing baked into the CAA to do this nicely from what I can see.

evereux, 2020-07-22:
I've created an example (in developement only atm) that expands a bit on this: https://github.com/evereux/pycatia/blob/developement/example_022.py

andrewellis55, 2020-07-23:
Works like a charm! Thanks!

closed 2020-07-23 by andrewellis55