evereux/pycatia #35: How to add the new instance in product?

tcrakshi, 2020-07-21:


tcrakshi, 2020-07-21:
How to add the new instance in product?
snippet will be more help

evereux, 2020-07-21:
Take a look at this: https://github.com/evereux/pycatia/blob/master/tests/create_source_products.py

tcrakshi, 2020-07-22:
hi,
i tried with above mentioned code, still its failed,
getting error like : com_error: (-2147352567, 'Exception occurred.', (0, 'CATIAProducts', 'The method AddComponent failed', None, 0, -2147467259), None)

Could you please help me in that

My code:

from pycatia import catia
documents = catia.documents
from pycatia.enumeration.enumeration_types import cat_constraint_type, cat_constraint_mode, cat_constraint_angle_sector, cat_work_mode_type

def loop(iProduct):
    root_Product = iProduct.products
    products = iProduct.get_products()
    for child_Product in products:
        root_Product.add_component(child_Product )

if __name__ == "__main__":
	document = catia.active_document
	product = document.product()
	# don't apply this at the function level, it only needs to be done once.
	product.apply_work_mode(cat_work_mode_type.index("DESIGN_MODE"))
	loop(product)

Error:

runfile('C:/Users/A337885/Desktop/PY/g.py', wdir='C:/Users/A337885/Desktop/PY')
Traceback (most recent call last):

  File "<ipython-input-41-877de5b8f8f6>", line 1, in <module>
    runfile('C:/Users/A337885/Desktop/PY/g.py', wdir='C:/Users/A337885/Desktop/PY')

  File "C:\Anaconda3\lib\site-packages\spyder_kernels\customize\spydercustomize.py", line 827, in runfile
    execfile(filename, namespace)

  File "C:\Anaconda3\lib\site-packages\spyder_kernels\customize\spydercustomize.py", line 110, in execfile
    exec(compile(f.read(), filename, 'exec'), namespace)

  File "C:/Users/A337885/Desktop/PY/g.py", line 16, in <module>
    loop(product)

  File "C:/Users/A337885/Desktop/PY/g.py", line 9, in loop
    root_Product.add_component(child_Product)

  File "C:\Anaconda3\lib\site-packages\pycatia\product_structure_interfaces\products.py", line 71, in add_component
    return Product(self.products.AddComponent(i_reference_product.com_object))

  File "<COMObject <unknown>>", line 2, in AddComponent

com_error: (-2147352567, 'Exception occurred.', (0, 'CATIAProducts', 'The method AddComponent failed', None, 0, -2147467259), None)

tcrakshi, 2020-07-22:
i got it. its working now

my Code:
from pycatia import catia
documents = catia.documents
from pycatia.enumeration.enumeration_types import cat_constraint_type, cat_constraint_mode, cat_constraint_angle_sector, cat_work_mode_type

def loop(iProduct):
    root_Product = iProduct.products
    products = iProduct.get_products()
    for child_Product in products:
        instance = child_Product.reference_product
        root_Product.add_component(instance)

if __name__ == "__main__":
	document = catia.active_document
	product = document.product()
	# don't apply this at the function level, it only needs to be done once.
	product.apply_work_mode(cat_work_mode_type.index("DESIGN_MODE"))
	loop(product)

evereux, 2020-07-22:
You still aren't formatting your code properly. Please do this in the future.

closed 2020-07-22 by evereux

tcrakshi, 2020-07-22:
while pasting in gituub or coding format in my editor?

i am new to github, i will learn in future

evereux, 2020-07-22:
That's fine but I've pointed you to links a few times showing where to read up on github formatting. There's a link at the bottom right of this box.

It takes time to support here and even more so when I can't read the code.