evereux/pycatia #36: CATDUA

tcrakshi, 2020-07-21:
Is it CATDUA interface exposed in pycatia?

evereux, 2020-07-21:
No. From what I understand there is no interface for CATDUA (it's a command line utility). It's possible to include it but lower priority right now.

tcrakshi, 2020-07-21:
Ok good to hear it's possible in pycatia. In most o to automotive organisation before releasing the catia data,  CATDUA perform is must. So if it is possible I request to you take this request on priority. .

evereux, 2020-07-21:
If you can organise funding for it I'll raise the priority.  😄

low-life-products, 2020-08-05:
I'll not be looking into doing this any time soon.

Kanfit, 2021-01-09:
Below can be useful:
[http://www.coe.org/p/fo/et/thread=17419](url)

tcrakshi, 2021-01-21:
above mentioned link, not fruitful.

Problem:
in most of the industry , in certain stage 3D cad data need to release - in terms of release process they need to do Cleaning of CAD data is also one of the requirement -   now CATDUA will come into picture. In this Stage Catia User (Designer) need to do CATDUA manually by going desk and and selecting CATDUA - Clean option. similarly they need to do hell lot of part  one by one.

Requirement:
for all opened document in Catia  - CATDUA should be done in one click of button or may be just calling on function.

tcrakshi, 2021-02-26:
Any update connected to CATDUA?

evereux, 2021-03-16:
> for all opened document in Catia - CATDUA should be done in one click of button or may be just calling on function.

This oversimplifies the task.

I'm aware of the typical release process and the requirement to clean CATIA data. No need to keep explaining that. 😄

However, to do this properly (not just satisfy your use case) is quite the task. For example, take a look at the error select options in the CATDUA tool. 

I just don't have the time right now. Maybe in the future. Maybe.

tcrakshi, 2021-03-16:
What exactly 'error select option'?
If it is cleaning all the error in data, irrespective of select , that would be the best.

evereux, 2021-03-16:
That would be best for you. That isn't incorporating CATDUA as a feature complete tool.

Perhaps you should look at setting up your own batch jobs.
http://catiadoc.free.fr/online/cfyugcatdua_C2/cfyugcatdua02.htm

tcrakshi, 2021-03-16:
Ya, currently we are following the above batch mode only. But problem is, all the user not aware and it need some knowledge. And also most of the time he want to run the CATDUA only selected document.  So user need to go desk and do it for required part only.

evereux, 2021-07-15:
No. Highly unlikely I'll do this this year if at all.

evereux, 2022-04-19:
I'm closing this issue as this isn't something that I can see ever being incorporated into pycatia.

closed 2022-04-19 by evereux