evereux/pycatia #38: How to add constrained  for product

tcrakshi, 2020-07-22:
1.how to add  Fix, offset, coinside,,,etc to specific product 
2. how to update specific product  constrained

evereux, 2020-07-22:
I don't think this was working properly and it's not obvious where to find it anyway. 

I had to use CATIA to record a macro and view the result in the visual code editor. This btw is a great to figure out how to do things. pycatia is very similar but more readable names.

I've changed that in the developement branch and added example_023. It will be included in the next release.

tcrakshi, 2020-07-22:
ok, thanks for information,
i have done it in VBA CATScript, if you required i can provide that code

evereux, 2020-07-24:
The latest version of pycatia should now have this working. See example_023.py for working example.

evereux, 2020-07-24:
Closing this issue as an example is provided (please also see sketch examples as they have constraints too).

closed 2020-07-24 by evereux

hosseintfr, 2023-09-28:
How can add a constraint (like coincidence of two planes of two parts) in a product by pycatia?
Please give me an example code.
Thanks
@evereux

hosseintfr, 2023-09-29:
@tcrakshi Do you know how can I run a Catia macro by python? can you please tell me the code you use?

hosseintfr, 2023-09-29:
@evereux What is exaple_023.py? I couldn't find that. I want to learn how to define constraints in catia product by pycatia but I couldn't. It's very hard without an example. Please help me.

TalentPoolAcademy, 2023-09-29:
CATIA Macro (VBA code) runs through Python I didn't try. But by using pycatia you can replicate your macro or you can develop how you want it and it needs to write the code in Python.

evereux, 2023-09-29:
@hosseintfr see the [readme](https://github.com/evereux/pycatia). You'll see links to the examples and documentation such as it is. The examples have since been renamed but see the section in the examples labelled [constraints](https://pycatia.readthedocs.io/en/latest/examples.html#example-constraints).

This issue has been closed. If you are having a specific problem please start a new issue but do read the requirements first as if they're followed it allows me to much easier help you.