evereux/pycatia #39: is it possible to check  which command is running on active catia window?

tcrakshi, 2020-07-22:
1. is it possible to check which command is running on catia?
2. is it possible to know active  msg box or dialog box info in active window catia?

evereux, 2020-07-24:
I don't understand this question. Can you provide more information on what it is you're trying to achieve?

tcrakshi, 2020-07-24:
Let us consider, some catia command is active, respecive window also be there. so my question is it, is to possible to fetch that information in pycatia. 
Example :
Point command is active in catia
Expectation in pycatia:  info : Point command is active

evereux, 2020-07-24:
I don't know. Is it possible to do that using the CAA interface?

tcrakshi, 2020-07-24:
I think yes, its possible in CAA

evereux, 2020-07-25:
I'm still struggling to understand the question and usage scenario to be honest.

tcrakshi, 2020-09-03:
Example:
1. let assume command lunched - through Code  CATIA.Startcommad ("xxxxx")
2. once command lunched  - command have a warning message with button yes and NO
3. so until and unless pressing yes or no button in catia,  next line of code will not excute.  

so now here i want press those yes or no button through command, is it possible to do that?

evereux, 2020-09-03:
Think I can understand now. 

You can disable file alerts.

```python
from pycatia import catia
caa = catia()
caa.display_file_alerts = False
```

Does this do what you need?

evereux, 2021-03-15:
I'm closing this as there has been no feedback for several months. If this is still an issue please open a new issue.

closed 2021-03-15 by evereux