evereux/pycatia #43: select_element2  and  select_element3   object error

tcrakshi, 2020-07-26:
Hi,
when i am using select_element2  and  select_element3 object in pycatia i am getting error like below mentioned.
TypeError: Objects for SAFEARRAYS must be sequences (of sequences), or a buffer object
any help is appreciated.
i hope this time , i have formatted the code, if it is not showing formatted please let me know.

**my code pasted below:**

`from pycatia import catia  
documents = catia.documents
rootProduct = 0
try:
    documents1 = catia.active_document
    rootProduct = documents1.product()
except:
    print ("Product is not open")
    sys.exit()
else:
    print("Product Documnet found")
Object_Filter = ("Product")
catia.active_document.selection.clear
selectionSet = catia.active_document.selection
status = selectionSet.select_element2(Object_Filter , "Select Product" , True)
#status = selectionSet.select_element3(Object_Filter ,"Select Product" , True , 1 , True)`

**Error**
`Traceback (most recent call last):

  File "<ipython-input-5-60de4e074a48>", line 1, in <module>
    runfile('C:/Users/A337885/Desktop/PY/untitled1.py', wdir='C:/Users/A337885/Desktop/PY')

  File "C:\Anaconda3\lib\site-packages\spyder_kernels\customize\spydercustomize.py", line 827, in runfile
    execfile(filename, namespace)

  File "C:\Anaconda3\lib\site-packages\spyder_kernels\customize\spydercustomize.py", line 110, in execfile
    exec(compile(f.read(), filename, 'exec'), namespace)

  File "C:/Users/A337885/Desktop/PY/untitled1.py", line 17, in <module>
    status = selectionSet.select_element2(Object_Filter , "Select Product" , True)

  File "C:\Anaconda3\lib\site-packages\pycatia\in_interfaces\selection.py", line 1623, in select_element2
    i_object_selection_before_command_use_possibility)

  File "<COMObject <unknown>>", line 4, in SelectElement2

TypeError: Objects for SAFEARRAYS must be sequences (of sequences), or a buffer object.`

evereux, 2020-07-26:
> i hope this time , i have formatted the code, if it is not showing formatted please let me know.

You haven't. Use the preview tab before posting to correct any errors.

```
    ```python
        print("hello world!")
    ```
```

will give you

```python
print("hello world!")
```

evereux, 2020-07-26:
Also, take a look at https://www.python.org/dev/peps/pep-0008/ to learn a little about the python language styling code guide lines.

evereux, 2020-07-26:
```python
Object_Filter = ("Product")
```

This is the problem and it caught me out too! The object type you have created here is a ``str`` and not a ``tuple``.  Try this:

```python
object_filter = ("Product",)
```

😃

tcrakshi, 2020-07-27:
Thanks for support, i will check and give feedback

tcrakshi, 2020-07-27:
Yes, it's working now.
Need to improve my python coding, anyway thanks for your support

closed 2020-07-27 by tcrakshi

evereux, 2020-07-27:
No shame is missing that one I don't think.

In python if you want to check type you can do:

```python
>>> a = ("this is a string")
>>> b = ("this is a tuple",)
>>> type(a)
<type 'str'>
>>> type(b)
<type 'tuple'>

```