evereux/pycatia #44: any object to specific object in selection

tcrakshi, 2020-07-27:
Hi, 
my requirement is to apply the design mode only for selected object. 

Error:
**AttributeError: 'AnyObject' object has no attribute 'apply_work_mode'**

Can i get help regarding this?

My Code:
```pyhon code
from pycatia import catia  
documents = catia.active_document
rootProduct = 0
try:
    rootProduct = documents.product()
except:
    print ("Product is not open")
    sys.exit()
else:
    print("Product Documnet found")

ProductSelection = []
object_filter = ("Product",)
selectionSet = catia.active_document.selection
status = selectionSet.select_element3(object_filter,"Select Product" , False , 2 , False)
if status == "Normal":
    selectionCount =  catia.active_document.selection.count
    selected_product = documents.product()
    for i in range(0 , selectionCount):
        selected_product = catia.active_document.selection.item(i+1).value
        ProductSelection.append(selected_product)
        
    catia.active_document.selection.clear()      
    for i in range(0 , len(ProductSelection)):
        ## For each selected product need to apply design Mode
        product = ProductSelection[i]
        product.apply_work_mode(cat_work_mode_type.index("DESIGN_MODE"))
```

**error with Traceback**
``` python error
Traceback (most recent call last):

  File "<ipython-input-33-60de4e074a48>", line 1, in <module>
    runfile('C:/Users/A337885/Desktop/PY/untitled1.py', wdir='C:/Users/A337885/Desktop/PY')

  File "C:\Anaconda3\lib\site-packages\spyder_kernels\customize\spydercustomize.py", line 827, in runfile
    execfile(filename, namespace)

  File "C:\Anaconda3\lib\site-packages\spyder_kernels\customize\spydercustomize.py", line 110, in execfile
    exec(compile(f.read(), filename, 'exec'), namespace)

  File "C:/Users/A337885/Desktop/PY/untitled1.py", line 27, in <module>
    product.apply_work_mode(cat_work_mode_type.index("DESIGN_MODE"))

AttributeError: 'AnyObject' object has no attribute 'apply_work_mode'

```

evereux, 2020-07-27:
Beautifully formatted .. nearly. 😉  👍 For python code add the word python after the opening three back ticks. See: https://github.com/evereux/pycatia/issues/43#issuecomment-663962049 No biggie though as I can now read it. 😄 

I see you've applied my suggestion here: https://github.com/evereux/pycatia/issues/43#issuecomment-663964831 If a issue is resolved please let me know and close the issue.

A question was asked last week that covers this same concept. Take a look at this example: https://github.com/evereux/pycatia/blob/developement/example_022.py The solution for you will differ of course but the concept is exactly the same. Basically you should test if selection is a product then create an instance of it. Once done you can then [apply_work_mode](https://pycatia.readthedocs.io/en/latest/api/pycatia/product_structure_interfaces/product.html#pycatia.product_structure_interfaces.product.Product.apply_work_mode).

tcrakshi, 2020-07-28:
Hi, 
i followed above mentioned example, its working now. Thanks for support

evereux, 2020-07-28:
Excellent. Glad you got it working! :+1:

closed 2020-07-28 by evereux

tcrakshi, 2021-09-17:
https://github.com/evereux/pycatia/blob/developement/example_022.py

its saying page not found, may i know where this example are moved?

tcrakshi, 2021-09-17:
''' Python
caa.active_document.selection.clear()
        object_filter = ("Product",)
        selectionSet = caa.active_document.selection
        status = selectionSet.select_element2(object_filter,"Select GAE Point Part" , False )
        if status == "Normal":
#            print(selectionSet.name)
            if selectionSet.item(1).type == "Product":
                Sel_object = selectionSet.item(1).value
                print(Sel_object.name)
                product_object = Sel_object.com_object
                GAE_Product = product(product_object)

tcrakshi, 2021-09-17:
'''Python
    try:
#        product1 = document.product()
        caa.active_document.selection.clear()
        object_filter = ("Product",)
        selectionSet = caa.active_document.selection
        status = selectionSet.select_element2(object_filter,"Select GAE Point Part" , False )
        if status == "Normal":
#            print(selectionSet.name)
            if selectionSet.item(1).type == "Product":
                Sel_object = selectionSet.item(1).value
                print(Sel_object.name)
                product_object = Sel_object.com_object
                GAE_Product = product(product_object)
#                print(GAE_Product.name)

'''

tcrakshi, 2021-09-17:
``` python Code
from pycatia import catia
import sys, time, os
from pycatia.enumeration.enumeration_types import cat_work_mode_type
from pycatia.product_structure_interfaces import product

caa = catia()
document = caa.active_document
product1 = document.product

caa.active_document.selection.clear()
object_filter = ("Product",)
selectionSet = caa.active_document.selection
status = selectionSet.select_element2(object_filter,"Select GAE Point Part" , False )
if status == "Normal":
    if selectionSet.item(1).type == "Product":
        Sel_object = selectionSet.item(1).value
        product_object = product(Sel_object.com_object)
        print (product_object.name)

```

tcrakshi, 2021-09-17:
i Found Answer 

it is Suppose same as below
```
product_object = product.Product(Sel_object.com_object)
```