evereux/pycatia #47: It's better to set 'Application.display_file_alters = False' when set 'overwrite=True' in document.save_to' method

Tian-Jionglu, 2020-08-12:
Hi,
here is a script I wrote to use in production.

### Task:
To change all the CATDrawings in a specific project. The client wants to change former company logo in border frame into new company name. The former logo is drawn with Polylines and filled with Area Fill. The border frame is placed in first detail view in the first detail sheet. All drawings is A0 frame and plus.

### Environment:
- Windows OS version: 10
- CATIA version: V5 R19
- Python version: 3.8.3
- pycatia version: 0.4.0

code:
```
import os
import glob
from pycatia import CATIADocHandler
from pycatia import catia

# set display_file_alerts = False, to avoid confirming overwrite
caa = catia()
caa.display_file_alerts = False

# get all the CATDrawings in the input 
directory = r'my directory'
file_pattern = os.path.join(directory, 'input', '*.CATDrawing')
files = glob.glob(file_pattern)

for file in files:
    with CATIADocHandler(file) as caa:
        # get the document, sheet and view object that would be used
        document = caa.document
        drawing = document.drawing_root()
        sheets = drawing.sheets
        sheet = sheets[1]
        views = sheet.views
        view = views[2]

        # since I can't find method to get polyline objects, I tried search method to get them and delete
        sel = document.selection
        sel.clear
        sel.add(view)
        sel.search("Drafting.Area Fill, sel")
        try:
            sel.delete()
        except:
            pass

        sel.clear
        sel.add(view)
        sel.search("Drafting.Polyline, sel")
        try:
            sel.delete()
        except:
            pass
        
        # add the new company name
        texts = view.texts
        text = texts.add('New Company Name', 20, 841/2)
        text.anchor_position = 6
        text.angle = 90
        text.name = 'Text_Corporation_Name'
        text.text_properties.font_size = 10
        text.set_font_name = 'Arial'
        sheet.force_update()

        new_file = os.path.join(directory, 'output', document.name)
        document.save_as(new_file, overwrite=True)
```
When debugging the code, the output directory was filled with some files. I have to click OK for each file to overwrite it. It's annoying.
When `overwrite=True` is set, I think the user already knows what would happen then. It would be better to set `Application.display_file_alters = False` than raise a warning to remind "Click OK" in command window.

Besides, I can't find a proper method the get the 2DPolyline object in CATDrawing. Anyone has good idea?

evereux, 2020-08-12:
> When debugging the code, the output directory was filled with some files. I have to click OK for each file to overwrite it. It's annoying.
When overwrite=True is set, I think the user already knows what would happen then. It would be better to set Application.display_file_alters = False than raise a warning to remind "Click OK" in command window.

Agreed. I'll make that update or you can submit a pull request. :+1: 

> I can't find a proper method to get the 2DPolyline object in CATDrawing. Anyone has good idea?

I can't find the tool itself in CATIA V5 and can't even ever remember using that apart from during my training course a long time back. Unless I'm missing something (entirely likely) I don't think I have access to this license right now. I'll investigate this further.

evereux, 2020-08-14:
> Besides, I can't find a proper method to get the 2DPolyline object in CATDrawing. Anyone has good idea?

Can you confirm that there is definitely such a thing in a CATDrawing? Whilst in a CATDrawing can you create one whilst recording a macro and post that here please?

Tian-Jionglu, 2020-08-15:
It's pity I also find no method to draw a ployline in CATDrawing directly.
In example above, the polylines are generated from exported files. The client designed his logo in other software and then imported to CATDrawing.
In CATIA, when we export a section to CATPart or CATDrawing, normally a polyline will we get. I made an example in this method. But I don't know how to  show it in github. I'm quite new here...
I checked the V5Automation Document, it only shows few information about this type of object. It is mentioned in `DrawingViewGenerativeBehavior (Object)` and `CatDrawingViewType (Enumeration)`. It seems polylines are used to define profile only.
Maybe it's not important to most users. :)

evereux, 2020-08-15:
> Agreed. I'll make that update or you can submit a pull request. 👍

I've now made the change to the development branch. Thanks for the suggestion!

Tian-Jionglu, 2020-08-18:
I've submitted a pull request to improve this example better.
Each time the function create_logger() was called, one more handler was added to Logger object. As a result, the displays increased for each implementation.
Run the code below to show the problem:
```
from pycatia import catia
caa = catia()
for i in range(3):
    caa.logger.info('Hi')
```
It is expected 3 infos to display, but you would see 6 (1+2+3) infos instead.

evereux, 2020-08-18:
Yeah I was aware of that quirk. :) Meant to take a look at that sometime.

Thank you.

evereux, 2020-09-26:
The original reason for the issue raised here is now resolved.

However, the polyline issue is not, so please do create a new issue if you so wish.

closed 2020-09-26 by evereux