evereux/pycatia #48: Is it possible to get list of files from Missing Files dialog on open?

imchaotic, 2020-08-17:
Currently if I open a CATProduct that is missing some components, I get a popup asking me to find those files.

Is it possible to get that list of files from the dialog window?

evereux, 2020-08-18:
I don't know if that functionality is built into the CAA API. But, each Product has a [path()](https://github.com/evereux/pycatia/blob/afa0fa374728080b33410cacaac6077c39b2464c/pycatia/product_structure_interfaces/product.py#L1345) method that returns a [Path](https://docs.python.org/3/library/pathlib.html) object. You could use that to see if the document [exists](https://docs.python.org/3/library/pathlib.html#pathlib.Path.exists).

evereux, 2020-09-03:
Closing this as there has been no response.

closed 2020-09-03 by evereux