evereux/pycatia #51: Add an existing part to a product

dastaninho, 2020-09-03:
Is it possible to insert an existing part to a product with this version of pycatia ?
Thanks

dastaninho, 2020-09-07:
I typed your code and got this 
File "U:\script\untitled2.py", line 15, in <module>
    part_product = part_doc.product()

  File "C:\Users\SEN5682\AppData\Local\conda\conda\envs\ML\lib\site-packages\pycatia\in_interfaces\document.py", line 727, in product
    return Product(self.document.Product)

AttributeError: 'Document' object has no attribute 'Product'

because part_doc became Document type and not Documents

Also, I couldn't find any examples that show how to do it.
Can you provide me some script please ?

Tian-Jionglu, 2020-09-08:
Hi,
this problem would happen with version 0.4.0. See issue #50 
two solutions:
1. try the newest code in `developement` branch, I think @evereux run the script in this enviroment.
2. try the code below with version 0.4.0.
```
from pycatia import catia

caa = catia()
documents = caa.documents

documents.add('Part')
part_doc = caa.active_document
part_product = part_doc.product()
part_product.part_number = "Wheel"

documents.add('Product')
product_doc = caa.active_document
product = product_doc.product()
product.part_number = "Car"

product.products.add_component(part_product)
```

evereux, 2020-09-10:
@Tian-Jionglu is right. :+1:  This example I provided was done in the developement branch. :-1: 

pycatia is now at 0.4.1 so if you upgrade the code I provided above should work.

evereux, 2020-09-25:
I'm closing this issue as there has been no feedback from OP and as far as I can tell the issue is resolved.

closed 2020-09-25 by evereux

hosseintfr, 2023-09-27:
I solved this problem and my code is working with pycatia 0.6.0:

from pycatia import catia
from pycatia.product_structure_interfaces.product_document import ProductDocument
from pycatia.mec_mod_interfaces.part_document import PartDocument

caa = catia()
documents = caa.documents

documents.add('Part')
part_doc = PartDocument(caa.active_document.com_object)
part_product = part_doc.product
part_product.part_number = "Wheel"

documents.add('Product')
product_doc = ProductDocument(caa.active_document.com_object)
product = product_doc.product
product.part_number = "Car"

product.products.add_component(part_product)

hosseintfr, 2023-09-27:
Also, for importing an existing component to product in Catia by pycatia, we can use this code (use your part destination instead of "D:\\New_folder\\Part1.CATPart"):

from pycatia import catia
from pycatia.product_structure_interfaces.product_document import ProductDocument
from pycatia.mec_mod_interfaces.part_document import PartDocument

caa = catia()
documents = caa.documents

documents.open(r"D:\\New_folder\\Part1.CATPart")
part_doc = PartDocument(caa.active_document.com_object)
part_product = part_doc.product
part_product.part_number = "Wheel"

documents.add('Product')
product_doc = ProductDocument(caa.active_document.com_object)
product = product_doc.product
product.part_number = "Car"

product.products.add_component(part_product)