evereux/pycatia #57: How to take screenshots from different directions

j-bau, 2020-10-01:
For documentation purpose I want to take screenshots from different directions. The screenshot itself is already working. Where I got stuck is the rotation of my viewpoint. 

```
from pycatia import catia
import pycatia.in_interfaces.viewpoint_3d as vp3d

caa = catia()
caa.documents.open('dummy.CATPart')

# create screenshot from iso view
caa.active_window.active_viewer.reframe()
caa.active_window.active_viewer.zoom_in()
caa.active_window.active_viewer.capture_to_file(5, 'iso.jpeg')

# create screenshot from x view
x_up = [1,0,0]
cur_vp3d = vp3d.Viewpoint3D(caa.active_window.active_viewer.com_object)
cur_vp3d.put_sight_direction(x_up)
caa.active_window.active_viewer.reframe()
caa.active_window.active_viewer.zoom_in()
caa.active_window.active_viewer.capture_to_file(5, 'x_up.jpeg')

```

Thank you

evereux, 2020-10-02:
I'd never used this before. Interesting!

You were very close. Take a look at this:

```python3

from pycatia import catia
from pycatia.in_interfaces.viewer_3d import Viewer3D

caa = catia()

document = caa.active_document
active_window = caa.active_window
active_viewer = active_window.active_viewer

x_up = (1, 0, 0)
view_point_3d = Viewer3D(active_viewer.com_object).viewpoint_3d
view_point_3d.put_sight_direction(x_up)
active_viewer.reframe()
active_viewer.zoom_in()```

evereux, 2020-10-04:
In the current development branch I've made some changes to the ViewPoint3D methods so the get methods should work.

I've also added an example that shows how to open all catparts in a folder and take screenshots for the views ISO, X, Y and Z as I thought it could be a useful reference. However, it'll only work with the development branch and 0.4.3 when it's released. (hopefully later today)

https://github.com/evereux/pycatia/blob/developement/example_027.py

j-bau, 2020-10-05:
Thank you very much! That is exactly what I was looking for.

I was able to sucessfully test your example with v0.4.3

Some minor additions to 'example_027.py':

- The export filename should be jpeg not png, because of catCaptureFormatJPEG

- The background color reset gets the wrong tuple
```
# reset background colour.
active_viewer.put_background_color(background_colour)
```
- Specification tree is not turned back on again
```
# turn specification tree back on 
specs_and_geom.layout = cat_specs_and_geom_window_layout.index("catWindowSpecsAndGeom")
```
Keep up the good work!

evereux, 2020-10-05:
Thanks for your feedback!

> The export filename should be jpeg not png, because of catCaptureFormatJPEG

School boy error! I wrote that bit before I realised PNG wasn't an option. :hand_over_mouth:  I'll fix that. 

I'll also get the other issues fixed later today.

Cheers

evereux, 2020-10-05:
> ```
> # turn specification tree back on 
> specs_and_geom.layout = cat_specs_and_geom_window_layout.index("catWindowSpecsAndGeom")
> ```

Can you confirm this is the case? This seems to work OK here. If so please report CATIA Version.

I've updated the other items mentioned. 👍

j-bau, 2020-10-06:
Sorry, there was a mistake on my side. The code works as intended. Thanks again for your help!

closed 2020-10-06 by j-bau