evereux/pycatia #63: [FEATURE REQUEST] Activate terminal node method.

sfp818, 2020-11-06:
on example 11 about acvivate terminal node,
I found some problem
or I am not sure this code is working properly.

because my product catia file size is too heavy, I usually use 'Do not activate default shape' option.

When I load product data first,
any product(cgr) data can't be seen,

even though loading is sucessful and there's no error in your example code.

In this condition(my option set up),
I really want to know how to activate terminal node after loading data by using python.
* my product consist of all cgr file(no parts)

[example11 in code]
from pycatia import catia from pycatia.enumeration.enumeration_types import cat_work_mode_type from pycatia.product_structure_interfaces.product import Product caa = catia() documents = caa.documents documents.open(r'D:\CATIA\test.CATProduct') document = caa.active_document product = document.product() product.apply_work_mode(cat_work_mode_type.index("DESIGN_MODE")) Product.activate_terminal_node(product.get_products())

evereux, 2020-11-08:
Your code is hard to read. Before posting please [read this](https://github.com/evereux/pycatia/issues/28#issue-660715005).

```python

from pycatia import catia 
from pycatia.enumeration.enumeration_types import cat_work_mode_type 
from pycatia.product_structure_interfaces.product import Product 

caa = catia() 
documents = caa.documents 
documents.open(r'D:\CATIA\test.CATProduct') 
document = caa.active_document 
product = document.product() 
product.apply_work_mode(cat_work_mode_type.index("DESIGN_MODE")) 
Product.activate_terminal_node(product.get_products())```

I can't see any problem here. Either with regular Product or Product with CGRs.

Which version of CATIA V5 are you using?

evereux, 2020-12-27:
Closing as OP hasn't provided further feedback.

If more information can be provided please re-open or start another ticket.

closed 2020-12-27 by evereux