evereux/pycatia #69: Iterators in pycatia

Tian-Jionglu, 2020-12-03:
When working on PR #66, some questions come into my mind:
1. pycatia use `__getitem__` method to make container iterable, but this method seems to be a legacy one. Instead, `__iter__` is a modern one. Is there some reason to choose `__getitem__`? See reference
    - https://stackoverflow.com/questions/926574/why-does-defining-getitem-on-a-class-make-it-iterable-in-python
    - https://stackoverflow.com/questions/20551042/whats-the-difference-between-iter-and-getitem
    - https://www.python.org/dev/peps/pep-0234/
2. Containers like *Products*, which inherit from *Collection*, could iterate well without defining `__getitem__` again in sub_class. It seems redundant here.

Tian-Jionglu, 2020-12-04:
After a test, I think use `__iter__` is a better choice. So there is PR #70.

About the test:
1. run the code below to create *CATPart* with 10000 points (**be careful that this code would last long**)
```
from pycatia import catia
caa = catia()
documents = caa.documents
document = documents.add("Part")
product = document.product()
product.part_number = "Points_Demo"
part = document.part()
hybrid_bodies = part.hybrid_bodies
hybrid_body = hybrid_bodies.add()
hybrid_body.name = "Points"
hsf = part.hybrid_shape_factory
for y in range(100):
    for x in range(100):
        point = hsf.add_new_point_coord(x, y, 0)
        hybrid_body.append_hybrid_shape(point)
part.update()
```
2. then run the code below to *Selection Iterator* either with PR #66  or #70
```
from pycatia import catia
import time

caa = catia()
document = caa.active_document
selection = document.selection
selection.search("type=Point, all")

start_time = time.time()
for selected_point in selection:
    pass
end_time = time.time()
print(end_time - start_time)
selection.clear()
```
The result shows `__iter__` method is faster than `__getitem__`. On my computer, it was about 9.8s and 11.5s.

evereux, 2020-12-28:
There was no reason for using __getitem_ over __iter__ and I was unaware of the histories.

I've updated all the methods that had __getitem__ so that they now use __iter__.

Tian-Jionglu, 2020-12-29:
Commit 464bbfe to fix bug in 3e57b3c.
`self.child_object` means object `SelectedElement` in this case.
`SelectedElement(self.child_object(self.com_object.item(i + 1)))` leads to error.

Besides, as far as I know, what method `__iter__` returns is exact `Iterator`.
We can get a `true` result with the code below.
```
from collections import Iterable
isinstance(<container in pycatia>, Iterable)  #<container in pycatia> represents any container with __iter__ method
```

Tian-Jionglu, 2021-01-27:
It topic is basically done.

I will start a new topic to discuss about the point below.
>     2. Containers like _Products_, which inherit from _Collection_, could iterate well without defining `__getitem__` again in sub_class. It seems redundant here.

closed 2021-01-27 by Tian-Jionglu

evereux, 2021-01-27:
Removing that method broke my tests when getting item by index (slicing). Please ensure any changes pass the test suite.