evereux/pycatia #71: Impove the behavior of broken links [FEATURE REQUEST]

Tian-Jionglu, 2020-12-07:
Suppose that we opened a root product, which has 5 sub products. And one of the sub product's link is broken.
Then we try the script below:
```
from pycatia import catia

caa = catia()
document = caa.active_document
product = document.product()
products = product.products
for sub_prd in products:
    print(sub_prd.file_name)
```
An **com_error** will occur when the process meet the broken link.
I think it's better to raise a "BrokenLinkErr" here. It would be more friendly for programmer to debug. (It take me a lot of time to find out this error when looping in a huge tree)

Or, if there is better solution to solve broken problems, it is appreciated.

evereux, 2020-12-27:
Thanks for the report!

I have submitted this patch: [Improve handling of Product.reference_product for broken links.](https://github.com/evereux/pycatia/commit/7efc7ecc2d113c37cfc94cee3bf2e5c0a85fb5cc).

evereux, 2020-12-27:
I'll be changing this patch as it has broken things ..

evereux, 2020-12-27:
OK. I think it's better now.

I didn't specifically raise a "BrokenLinkErr" as I don't know if other issues could also raise a com_error for this case.

Tian-Jionglu, 2021-01-13:
There are more methods need this improvement, like:
- [ ] Product.part_number
- [ ] Product.source
Should we do the same change one by one?
I'm sure Product.part_number is often used.

evereux, 2021-01-13:
Sure!

Tian-Jionglu, 2021-01-14:
Commit https://github.com/evereux/pycatia/pull/77/commits/6b45a5360f3a3eea3ab1d0ed332ae157bfebc5de to improve `Product.part_number` and `Product.soure` following this topic.

closed 2021-01-25 by Tian-Jionglu