evereux/pycatia #72: [FEATURE REQUEST] load Existing component

sfp818, 2020-12-26:
I usually use catia product file,

I really want to know how to add 'Exisiting component'

is it possible to insert 'existing component' on Product??

and one more question...
can I select product and activate by name?

I can only check product name
 - code : product.name
 
I just know to activate by index like this (product/part/cgr)
 [code]
product.get_products()[0].activate_default_shape()

**Is your feature request related to the CAA V5 Automation API?**
If yes, please describe where in the API documentation that feature is described.

**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

Tian-Jionglu, 2020-12-27:
Hi, 
about the first question, try the `Products.add_external_component`.
About the 2nd  question, try to create a mapping dict like `{<name>: <boolean>}` to control the switch of "active status".

Kanfit, 2020-12-27:
Yes, you can add existing component. act_prod_docs.AddComponentsFromFiles(components_list, "All"). In my case components_list is an array

sfp818, 2020-12-27:
Thanks all guys. 
plz check my code again. 
I don't know why it doesn't work properly.
To add exisiting components...

document = catia().active_document
product = document.product()
product.apply_work_mode(cat_work_mode_type.index("DESIGN_MODE"))

components_list = ('D:\\CATIA\\NQ5E_0105_1_cgr\\TEST.cgr', 'D:\\CATIA\\NQ5E_0105_1_cgr\\TEST2.cgr')
product.products.add_components_from_files(components_list, "ALL")

Tian-Jionglu, 2020-12-27:
This code normally works with CATPart or CATProduct, not cgr...

evereux, 2020-12-27:
I was able to add a CGR just fine running the following code:

```python

from pycatia import catia
from pycatia.enumeration.enumeration_types import cat_work_mode_type

document = catia().active_document
product = document.product()
product.apply_work_mode(cat_work_mode_type.index("DESIGN_MODE"))

components_list = (
    r'C:\Users\evereux\Desktop\catia_products\Part1.CATPart', r'C:\Users\evereux\Desktop\catia_products\box.cgr')
product.products.add_components_from_files(components_list, "*")
```

Note the addition of the `r` to the being of the file string definition.

sfp818, 2020-12-28:
thanks so much

thats perfect!!!!

closed 2020-12-28 by sfp818