evereux/pycatia #74: Points import from CSV file

Kanfit, 2021-01-07:
**Describe the bug**
During points to catia import (Example 5) following error occurs: FloatingPointError: Input {number} can not be converted to float(). Check csv data.

**To Reproduce**
Below will throw an exception

Point name | X coordinate | Y coordinate | Z coordinate
-- | -- | -- | --
PT1 | 1192 | 6862.9 | 177.3
PT2 | 0 | 6862.9 | 177.3
PT3 | -1192 | 6862.9 | 177.3
PT4 | -1192 | 6858.6 | 174.2
PT5 | -1192 | 6858.5 | 174.1
PT6 | -1192 | 6851.3 | 166.8
PT7 | -1192 | 6851.3 | 166.6
PT8 | -1192 | 6847.6 | 157.5
PT9 | -1192 | 6832.6 | 157.1
PT10 | 0 | 6832.6 | 157.1

Even if I convert numbers to 3 decimal places the same exception will occur

**Desktop (please complete the following information):**
 - OS: Windows 10
 - CATIA V5 version V5-6R2018
 - pycatia version pycatia 0.4.4

evereux, 2021-01-07:
Can you provide the file you are using please?

Kanfit, 2021-01-09:
I have attached a zip. Imports works perfect without header (columns name)
[Test.zip](https://github.com/evereux/pycatia/files/5790197/Test.zip)

evereux, 2021-02-08:
Example 5 explains what the format of the data should be.

I've added a note to example 5 to say that there should be no column header data.

closed 2021-02-08 by Kanfit