evereux/pycatia #78: Order points collection for output

Kanfit, 2021-01-10:
I try to output points coordinates in an order one by one. In the Catia tree points aren't ordered. Points are created in two steps. First step is end points creation. Second step is points on curve by ratio creation. Points are not ordered in the Catia tree. Now, for export points must be in an order. How can it be done?

evereux, 2021-01-10:
I'm not sure how you actually want the points ordered?

Any chance of labelled screenshots?

Kanfit, 2021-01-10:
Please see attached.
[image]

evereux, 2021-01-10:
Where are points 9 and 10? Why do you then jump to point 4?

Either way I can't think of any simple logic using the CATIA API to apply to this situation. Why is the order important? Can the points be ordered by coordinates somehow? This is something you could do with python itself.

Kanfit, 2021-01-10:
Sorry. Output should be: 3, 5, 6, 7, 8, 9, 10 and 4. I need that for laser projection. To understand requirement better, imagine you create a spline using these points. 
[image]

evereux, 2021-01-10:
Got it.

If it's possible I would change the way you create the points. Create the points on the line in one step using `Points And Plane Repetition` and check `With End Points`. I'm sure they'll be in the right order then.

Kanfit, 2021-01-10:
You are right. Will try that. Work on reverse engineering of VB6 app (For laser projection).

closed 2021-01-10 by Kanfit

RebelYoung, 2025-05-19:
@evereux Hello, is there `add_new_points_repetition` function in `0.8.2` ?

evereux, 2025-05-19:
@RebelYoung No. You would need to create a function that does that yourself. Dassault haven't exposed that from what I can see.