evereux/pycatia #80: [BUG]New product creation

Kanfit, 2021-01-18:
**Describe the bug**
New product creation when "Manual import" check box checked in the infrastructure options.

**To Reproduce**
Go to Options ->Infrastructure->Product Structure->Product structure tab and check Part Number manual input.
Add new product using Add method and try to set product name thru code.
Above will not work. Uncheck "Manual input" and you are good to go.

[https://stackoverflow.com/questions/31998761/create-new-product-in-catia-with-python](url)

To call VBA function in python works like a charm.
Any thought?

evereux, 2021-01-24:
Hmm, maybe embedding the VBA function into pycatia would be good for this use case. 

I tried looking to see if the setting can be checked / changed through the COM interface but I'm can't find it.

Tian-Jionglu, 2021-01-26:
If what I understand is right, this issue would be about setting controller.
#62 could be a reference.

Unfortunately, in my environment(V5R19), Tab "Options ->Infrastructure->Product Structure->Product structure" is not exposed by CATIA. It can be tested in later environment.

Generally, I would use "Dumps parameters values"(bottom left button in options window) to record a specific option tab. If it could record an useful VBScript, then setting control would be possible.

evereux, 2021-01-26:
I tried this myself the other day on R21 and it isn't exposed there either.

I have contacts who use much later releases. I'll get them to check.

Kanfit, 2021-01-27:
R28 - Isn't exposed

ptm-tm, 2023-10-11:
can u write simple code ? i check in some releases of catia V5R29 and maybe R30-31

deloarts, 2023-10-12:
R33 - not exposed