evereux/pycatia #81: Missing BasicComponent [FEATURE REQUEST]

Tian-Jionglu, 2021-01-19:
I'm trying to write a script to set the layer of elements in CATPart.
When I was trying to read the number of layers exist, I found the object BasicComponent is missing in pycatia.
`basic_component.py` is in **v5automation**, but not in pycatia application.
Is there reason that this object can't be exposed?

evereux, 2021-01-19:
There is nothing in the folder `CATAnalysisInterfaces` exposed within pycatia. There's no reason for this other than I was just tackling the low hanging fruit at the time.

I'll take a look at this but it might not be for a few weeks.

Tian-Jionglu, 2021-01-20:
I'll try to extent needed modules with v5automation.
Now I understand pycatia.readme more.
Your work are greatly appreciated! To make current pycatia work is really huge work.

evereux, 2021-01-20:
Thanks. :+1: 

I've created the branch `cat-analysis-interfaces`. I've simply dumped the contents of `v5automation-autogenerated\CATAnalysisInterfaces` into the folder `analysis_interfaces` for now. Of course feel free to patch / update the modules in line with the rest of pycatia. Thanks for any work you can do on this.

Tian-Jionglu, 2021-01-21:
I've create a project to organize this task. 
See https://github.com/users/Tian-Jionglu/projects/2
I need to reach the annotation API and layer API to reach my goal.
In branch [layer_and_annotation](https://github.com/Tian-Jionglu/pycatia/tree/layer_and_annotation) is annotation API partially added and could work as example below:
```
from pycatia import catia

caa = catia()
document = caa.active_document
part = document.part()
annotationSets = part.annotation_sets
print(annotationSets)
print(annotationSets.items())
```
But I am not aware of name `analysis_interfaces` before I renaming `CATTPSInterfaces` to `cat_tps_interfaces`.
Please give me a guide that would you want to change `cat_tps_interfaces` to `tps_interfaces`.

evereux, 2021-01-24:
Generally I removed the`cat` prefix from everything that has it as it doesn't add anything information wise. If you can think of a good reason to keep the `cat` prefix I'm happy for you to keep it.

Tian-Jionglu, 2021-01-25:
With pull request #82, my origin task is basically done.

Here is an example:
```
# Set the Annotations' layer to "Layer=100" in CATPart files
# This is a solution to manage visualization in a team

from pycatia import CATIADocHandler
import os

source_path = '**' # replace '**' to the directory containing source CATPart files

for root, dirs, files in os.walk(source_path):
    for file in files:
        if file[-4:] == 'Part':
            with CATIADocHandler(os.path.join(root, file)) as handler:
                document = handler.document
                part = document.part()
                selection = document.selection
                try:
                    annotationSets = part.annotation_sets
                    annotationSet = annotationSets.item(1)
                    annotations = annotationSet.annotations
                    for annotation in annotations:
                        selection.add(annotation)
                    vis_properties = selection.vis_properties
                    vis_properties.set_layer(0, 100)
                    document.save()
                except:
                    pass
```
I made a mistake at the first of this topic. `CATAnalysisInterfaces` seems to be about CAE solutions.

Tian-Jionglu, 2021-01-25:
**Notes for further maintenance of package `tps_interfaces`**
 - `Annotation` and `AnnotationSet` are basically added, but `UserSurface` not yet
 - modules below are not added yet
    {'annotation_set_transform_into_assembly_set.py',
     'assembly_annotation_sets.py',
     'dmu_tol_setting_att.py',
     'fta_infra_setting_att.py',
     'fta_setting_att.py',
     'non_semantic_datum.py',
     'non_semantic_datum_target.py',
     'non_semantic_dimension.py',
     'non_semantic_gdt.py',
     'semantic_gdt.py',
     'tolerance_per_unit_basis_restrictive_value.py',
     'tolerance_zone.py',
     'user_surface.py',
     'user_surfaces.py'}

evereux, 2021-02-08:
I don't want to selectively merge bits of the package tps_interfaces. Otherwise I'm concerned items will be lost / forgotten.

I've created a new branch `tps_interfaces`. Please submit your patches to that branch. 👍

Tian-Jionglu, 2021-02-13:
PR #82 has been modified to meet the request.

evereux, 2021-03-15:
Can this be closed now?

closed 2021-03-16 by Tian-Jionglu