evereux/pycatia #83: [DISCUSSION]About code reuse of Collection objects

Tian-Jionglu, 2021-01-27:
When discussing about issue #69, I see a possibility to improve the code reuse of `Collection` objects.
For an example, 

1. methods `__len__`, `__getitem__` and `__iter__` are defined in class `Collection`, then it is not needed to redefine in class `Products`
2. methods `get_item_by_index` and `get_item_by_name` in `Collection` are repeated by `item` in `Products`, it seems not good to define these two methods in a top level class

I'll try to make a pull request to present an example.

evereux, 2021-01-27:
There is a reason for how it currently is. Don't make a pull request yet. 

Let me explain properly another time why it is that way. That's not to say it can't be improved.

Tian-Jionglu, 2021-01-29:
I met some problem in testing. Some codes in tests seem out of date

`tests/create_source_drawing.py` as an example

```
import os
from pathlib import Path

from pycatia import catia as pia
from pycatia.enumeration.enumeration_types import cat_paper_size
from tests.create_source_parts import get_cat_part_measurable

test_files = Path("tests/cat_files")

source_cat_drawing = Path(os.getcwd(), test_files, "drawing.CATDrawing")

def create_cat_drawing():
    documents = pia.documents
    documents.add("Drawing")
    drawing_document = pia.active_document
    drawing_document.save_as(source_cat_drawing)
    drawing = drawing_document.drawing_root()
    sheets = drawing.sheets

    --snip--
```
Here  `documents = pia.documents` leads to error. `pycatia.catia` shoud be called since 0.3.6.
Should I change the codes?
It's my first time to use pytest framework. Maybe I'm wrong.

Tian-Jionglu, 2021-01-29:
Even with the problem above solved, there is still an error when creating the part.
Report as below. Please do a check.
```
tests\in\test_application.py:4: in <module>
    from tests.source_files import cat_part_measurable
tests\source_files.py:12: in <module>
    cat_drawing = get_cat_drawing()
tests\create_source_drawing.py:61: in get_cat_drawing
    create_cat_drawing()
tests\create_source_drawing.py:23: in create_cat_drawing
    documents.open(get_cat_part_measurable())
tests\create_source_parts.py:262: in get_cat_part_measurable
    create_cat_part_measurable(source_cat_part_measurable)
tests\create_source_parts.py:144: in create_cat_part_measurable
    con_line_1_start = constraints.add_bi_elt_cst(cat_constraint_type.index("catCstTypeOn"), line_1_2d.start_point,
pycatia\mec_mod_interfaces\constraints.py:137: in add_bi_elt_cst
    return Constraint(self.constraints.AddBiEltCst(i_cst_type, i_first_elem.com_object, i_second_elem.com_object))
<COMObject <unknown>>:2: in AddBiEltCst
    ???
E   pywintypes.com_error: (-2147352571, '类型不匹配。', None, 2)
```

evereux, 2021-01-29:
I'll take a look. I haven't run the tests with no files created for awhile. Cheers.

evereux, 2021-02-08:
I've updated the tests document creation and they now work for me from an empty `tests\cat_files` directory (empty of CATIA documents only).

Tian-Jionglu, 2021-02-14:
The test runs  still in failure when creating the `part_measurable.CATPart`.
The problem seems in `constraints.add_bi_elt_cst` in my environment (V5R19).
It runs well when I annotate the codes below in `tests/create_source_parts.py`.
```
    # constraints = sketch.constraints

    # con_line_1_start = constraints.add_bi_elt_cst(cat_constraint_type.index("catCstTypeOn"), line_1_2d.start_point,
    #                                               point_1)
    # con_line_1_start.mode = cat_constraint_mode.index("catCstModeDrivingDimension")
    # con_line_1_end = constraints.add_bi_elt_cst(cat_constraint_type.index("catCstTypeOn"), line_1_2d.end_point, point_2)
    # con_line_1_end.mode = cat_constraint_mode.index("catCstModeDrivingDimension")

    # con_line_2_start = constraints.add_bi_elt_cst(cat_constraint_type.index("catCstTypeOn"), line_2_2d.start_point,
    #                                               point_2)
    # con_line_2_start.mode = cat_constraint_mode.index("catCstModeDrivingDimension")
    # con_line_2_end = constraints.add_bi_elt_cst(cat_constraint_type.index("catCstTypeOn"), line_2_2d.end_point, point_3)
    # con_line_2_end.mode = cat_constraint_mode.index("catCstModeDrivingDimension")

    # con_line_3_start = constraints.add_bi_elt_cst(cat_constraint_type.index("catCstTypeOn"), line_3_2d.start_point,
    #                                               point_3)
    # con_line_3_start.mode = cat_constraint_mode.index("catCstModeDrivingDimension")
    # con_line_3_end = constraints.add_bi_elt_cst(cat_constraint_type.index("catCstTypeOn"), line_3_2d.end_point, point_4)
    # con_line_3_end.mode = cat_constraint_mode.index("catCstModeDrivingDimension")

    # con_line_4_start = constraints.add_bi_elt_cst(cat_constraint_type.index("catCstTypeOn"), line_4_2d.start_point,
    #                                               point_4)
    # con_line_4_start.mode = cat_constraint_mode.index("catCstModeDrivingDimension")
    # con_line_4_end = constraints.add_bi_elt_cst(cat_constraint_type.index("catCstTypeOn"), line_4_2d.end_point, point_1)
    # con_line_4_end.mode = cat_constraint_mode.index("catCstModeDrivingDimension")
```
Is it alright in your environment?

evereux, 2021-02-14:
Yeah. I just tried several times, each time deleting that part. The sketch is created a properly constrained.

Are you sure your branch is synced with the developement branch?

Tian-Jionglu, 2021-02-15:
I've solve this problem in branch https://github.com/Tian-Jionglu/pycatia/tree/fix_tests.

VB error -2147352571 seems only happen in low version of VB (or CATIA).
Methods like `constraints.add_bi_elt_cst` must strictly use `Reference` object as argument in low version CATIA (V5R19).

closed 2024-04-22 by evereux