evereux/pycatia #84: [FEATURE REQUEST] Rename selected elements using csv file

Kanfit, 2021-01-27:
There is a a nice method to import/export points using csv. Any methods to rename selected elements using csv file (just one column that is element name)? I can read csv line by line but selected elements always get last row value.

Tian-Jionglu, 2021-01-28:
Try this.
```
import csv
from pycatia import catia

caa = catia()
document = caa.active_document
part = document.part()
hybrid_bodies = part.hybrid_bodies
hybrid_body = hybrid_bodies.item('points')
points = hybrid_body.hybrid_shapes

with open('points.csv', 'r') as csv_file:
    csv_reader = csv.reader(csv_file)
    for row in csv_reader:
        try:
            point = points.item(row[0])
            point.name = row[1]
        except:
            pass
```

Supposed a CATPart is opened, and target points lay in a GeometricalSet named "points".
Code works with csv file below, named 'points.csv'. Header is better to be ignored.

current_name | target_name
-- | --
Point.1 | Point_A
Point.2 | Point_B
Point.3 | Point_C
Point.4 | Point_D
Point.5 | Point_E
Point.6 | Point_F
Point.7 | Point_G
Point.8 | Point_H
Point.9 | Point_I
Point.10 | Point_J
Point.11 | Point_K

Kanfit, 2021-01-28:
I have only one column that is a new name. Below code renames each curve several times and I end up with a last row value:
```
with open(csv_file, 'r') as names_file:
    names_reader = csv.reader(names_file)
    for row in names_reader:
        for i in range(curves.count):
            try:
                curve = curves.item(i + 1)
                curve.name = row[0]
            except Exception as err:
                print(err)
```
Any help would be highly appreciated.

Tian-Jionglu, 2021-01-28:
I think the problem is here. 
```
    for row in names_reader:
        for i in range(curves.count):
```
Try this:
```
with open(csv_file, 'r') as names_file:
    names_reader = csv.reader(names_file)
    for i, row in enumerate(names_reader):
        try:
            curve = curves.item(i + 1)
            curve.name = row[0]
        except Exception as err:
            print(err)
```

Kanfit, 2021-01-28:
Thank you. Works perfect and saves  a lot of time.

closed 2021-01-31 by Kanfit