evereux/pycatia #85: [BUG] active_document method fails

Kanfit, 2021-01-31:
**Describe the bug**
Some times on heavy operations active_document method fails.
I convert multi-body part to an assembly and method fails 
Sometimes fails after 3 bodies conversion sometimes after 10.

**Desktop (please complete the following information):**
 - OS: Windows 10
 - CATIA V5-6R2018
 - pycatia 0.4.4

**Additional context**
With 10 bodies method works alright.

Tian-Jionglu, 2021-02-01:
Hi, could you paste some code, or full stack trace of error.
It's hard to understand the problem... 😂

Kanfit, 2021-02-01:
Hi,

Omit kivy declarations.
Basically I copy each body, create new part perform some operations such as random color application and save part. Fails during MainBody declaration and only if I have a lot of bodies. With 10 bodies works perfect.

```
def part_bodies_to_assy():
	logger.AppLogger.app_logger()
	
	with open(app_settings + '\\Settings.json') as file:
		data = json.load(file)

# region Popup worker (Progress bar)
	progress_bar = PB()
	lb = WorkerLlb(
		text='',
		font_size=20,
		markup=True
	)
	box = Box(orientation='vertical')
	box.add_widget(progress_bar)
	box.add_widget(lb)

	worker = WorkerPopup(
		title='Part bodies to assy',
		background='../Assets/Images/MainBG.png',
		content=box,
		title_size=20,
		title_align='center',
		separator_height=3,
		auto_dismiss=False,
		size_hint=(.8, .5)
	)
	worker.open()
# endregion

	progress_bar.value = 0
	time.sleep(.5)

	ca_v5 = catia_v5_mng.CatV5Application().get_catia()
	act_doc = ca_v5.ActiveDocument
	part_doc = act_doc.Part
	part_rev = act_doc.Product.Revision
	prt_name = part_doc.Name
	bodies = part_doc.Bodies
	sel = act_doc.Selection
	input_path = str(act_doc.path + "\\")
	# Disable display refresh and Catia visibility

	sel.Clear()  # Clear selections
	# Force clear history. Catia may keep session in a history
	ca_v5.StartCommand(cat_v5_cmd.cmd('Cache'))
	# Warn user on a background process
	lb.text = '[b][color=7FFF00]Do not alter the operation! Running on the background.[/color][/b]'
	time.sleep(data['AppSettings']['Delay'][0]['MessagesStatusLongDelay'])
	ca_visibility = data['AppSettings']['Catia'][0]['Visibility']
	ca_refresh_display = data['AppSettings']['Catia'][0]['RefreshDisplay']
	ca_v5.RefreshDisplay = ca_refresh_display
	ca_v5.Visible = ca_visibility
	
	lb.text = 'Renaming bodies'
	time.sleep(data['AppSettings']['Delay'][0]['MessagesStatusDelay'])

	# Rename all part bodies with catia_worker part number followed by body index (add dash No.)
	# TODO To check if body is hidden. Skip hidden bodies in a loop
	for bodies_to_rename in range(bodies.Count):
		bodies_count = bodies.Count
		progress_bar.max = bodies_count
		part_body = bodies.Item(bodies_to_rename + 1)
		progress_bar.value = bodies_to_rename + 1
[... 194 lines cut ...]
		# Set reference
		reference = product.create_reference_from_name(component_name)
		# Add fix constraints
		constraints.add_mono_elt_cst(cat_constraint_type.index("catCstTypeReference"), reference)
	# Clear selection, search for constraints and hide them
	sel.clear()
	sel.search(
		"((((((((CATProductSearch.MfConstraint + CATStFreeStyleSearch.MfConstraint) "
		"+ CATAsmSearch.MfConstraint) + CAT2DLSearch.MfConstraint) + CATSketchSearch.MfConstraint) "
		"+ CATDrwSearch.MfConstraint) + CATPrtSearch.MfConstraint) + CATSpdSearch.MfConstraint) "
		"& Visibility=Visible),all"
	)
	vis_props.set_show(1)
	sel.clear()
	caa.start_command(cat_v5_cmd.cmd('FitAll'))
	# Display Catia V5 window
	# Delays to notify on completion
	lb.text = str("Done")
	time.sleep(data['AppSettings']['Delay'][0]['MessagesStatusDelay'])
	lb.text = str("Exit application and save assembly file")
	time.sleep(data['AppSettings']['Delay'][0]['MessagesStatusLongDelay'])
	worker.close_worker()
	caa.refresh_display = True
	caa.visible = True
```

Tian-Jionglu, 2021-02-02:
The video shows a successful running, right? 
Line 106 and 107 seem not safe, when there are bodies named the same. Maybe you can have a check.
```
                temp_act_doc = ca_v5.ActiveDocument
                temp_part_doc.Product.PartNumber = bodies.Item(i + 1).Name
```

Kanfit, 2021-02-02:
Yes, video shows successful run. I rename bodies prior to copy with a running number, so no issues with file name.

evereux, 2021-02-08:
Can you post some code we can run that reproduces the problem? I don't want to try and figure out if the problem lies with another package starting a process that the COM interface isn't ready for.

Kanfit, 2021-02-08:
I will clean a project from kivy code and will post

evereux, 2021-02-08:
Thanks. 

It was interesting to see pycatia being used with a KIVY front end btw. It's nice to see the project being used in the wild like this. 

I had plans on creating a GUI (KIVY maybe?) front end for a Q-Check type application. But, that's not going to happen for a few months at least. It's all work and no play at the moment. 😃

Kanfit, 2021-02-08:
I like kivy and it's easy to use and more over it's a cross platform, so I can cover desktop and mobile as well. Packaging is easy once you nail it. I upload new vids to youtube from time to time. Feel free to contact me.

evereux, 2021-03-15:
Can we get a stripped down, pycatia only version of the code so we may reproduce this issue?

evereux, 2021-07-25:
There has been no feedback for some months now so will close this issue as I can't see evidence of a problem with pycatia itself.

I'll open issue again if a pure pycatia code sample can be provided.

closed 2021-07-25 by evereux