evereux/pycatia #86: About example 016

hansen-cpu, 2021-02-14:
Windows 10 64位
CATIA version: V5R21
Python version: 3.8.5
pycatia version: 0.4.4

I'm still learning how to use pycatia. However, example 016 just doesn't work. The problem is on the line 49, 
"abs_axis = geom_elements.item("AbsoluteAxis")".
It says "pywintypes.com_error: (-2147352567, '发生意外。', (0, 'CATIAGeometricElements', '方法 Item 失败', None, 0, -2147352573), None)"
I don't know how to fix it.

evereux, 2021-02-14:
That works ok for me.
What's outputted to the terminal when you add the following before that line.

```python
print(geom_elements[0])
```

I get

```
GeometricElement(name="AbsoluteAxis")
```

Tian-Jionglu, 2021-02-15:
@hansen-cpu, normally this is because there is no "AbsoluteAxis" in your CATPart Document.
Try to check **Options -> Infrastructure -> Part Infrastructure -> Part Document -> Create an axis system**, then create a new CATPart to run this example.

hansen-cpu, 2021-02-15:
Thank you very much. Your answers helped me a lot. Thanks.

hansen-cpu, 2021-02-15:
I got it. My CATIA is a Chinese translated version. As a result, there's no "AbsoluteAxis" in the geometric set. Instead, it's the Chinese translated version "绝对轴". Thank you very much.

Tian-Jionglu, 2021-02-15:
Is this example alright now with V5R21environment?

This example runs in failure in my V5R19 environment, because of `constraints.add_bi_elt_cst` method.
The same error also met in https://github.com/evereux/pycatia/issues/83#issuecomment-769600153.
I'm looking into this problem. If this example is alright now in V5R21, please give me a feedback.

hansen-cpu, 2021-02-15:
It's alright in V5R21. I think this problem is about language or translation.

Tian-Jionglu, 2021-02-15:
Thank you for feedback.

evereux, 2021-03-15:
Closing as issue is resolved.

closed 2021-03-15 by evereux