evereux/pycatia #87: [DISSCUSSION]How about using specific document to replace Document

Tian-Jionglu, 2021-02-19:
In branch [specific_document](https://github.com/Tian-Jionglu/pycatia/tree/specific_document), it shows a skill to let `active_document` method return a specific document type.

For example, run the code below,
```
from pycatia import catia
caa = catia()
document = caa.active_document
```
now it will return `PartDocument` or `ProductDocument`, instead of `Document`.

Why I want to open this issue?
At present, pycatia merged all methods from specific document (like ProductDocument, PartDocument) into class `Document`.
When I want to add `FunctionalDocument` into pycatia, I think class `Document` would become too huge if following the present strategy.

It's  only a demo at present, more codes need to be checked.
I hope you like it.

evereux, 2021-03-16:
Agreed and is something I should have done from the start. But, at the start it was a fairly simple project. 😄 

I've made those updates with some tweaks to development branch.

* version is now bumped to 0.5.0 as this is a breaking change.
* I've moved the types files to own folder.
* I added CATDrawing.
* All the tests and examples run ok.

Tian-Jionglu, 2021-03-16:
In development branch now contains pycatia 2 document.py.
1. pycatia/types/document.py
2. pycatia/in_interfaces/document.py

It looks weird. 😅

evereux, 2021-03-16:
It does. I'll change it.

evereux, 2021-07-25:
I think this can be closed now so will do so.

If there is more work to do I've no problem opening it again.

Cheers

closed 2021-07-25 by evereux