evereux/pycatia #89: [FEATURE REQUEST] Change setup.py to allow for wider range of dependency versions

andrewellis55, 2021-04-09:
**Is your feature request related to the CAA V5 Automation API?**
No

**Describe the solution you'd like**
When installing pycatia, my venv must contain the exact version of the depdancies
    'atomicwrites==1.2.1',
    'attrs==18.2.0',
    'colorama==0.4.0',
    'more-itertools==4.3.0',
    'pluggy==0.8.0',
    'py==1.7.0',
    'pywin32>=224'

Other packages that I use with pycatia are now requiring higher version of some of these tools. Is it possible to change any of these to >= rather than ==? I understand that normally in a requirements file == can be used since a specific setup is likely wanted for any CI that uses the file, but a setup file typically isn't as restrictive if there is no specific reason for the restriction. Is is possible to loose some of this restrictiveness in the setup?

evereux, 2021-04-21:
Agreed. I'll make the changes soon.

evereux, 2021-04-27:
The only requirement in that list for pycatia is actually pywin32. All the others as far as I can see are only used for testing and are install by pytest. So, I've removed them all.

I shall update pypi shortly.

andrewellis55, 2021-04-27:
Looks great, thanks so much!

closed 2021-04-27 by andrewellis55