evereux/pycatia #90: Table row size larger than it should be

ab0v3g4me, 2021-04-23:
I've got this simple script that creates a table  of size 9x7. The problem is that the row sizes are 7mm(6mm requested size) and i can't seem to lower them any more. I've tried this manually, with VBA, and now with pycatia and none seem to give me a row with size of 6mm. 
```python
from pycatia import catia

ca = catia()

documents = ca.documents
all_sheets = []

for d in documents:
    if d.full_name.endswith("CATDrawing"):
        for sheet in d.drawing_root().sheets:
            all_sheets.append(sheet)

current_sheet = all_sheets[0]

background_view = current_sheet.views.get_item_by_name('Background View')
background_view.activate()

tables = background_view.tables

test_table = tables.add(100, 100, 7, 9, 6, 20)

```

evereux, 2021-04-27:
The minimum row height is limited by your text height plus any padding(?). Try using a smaller font.

I've tweaked your script for the latest version 0.5.0 for anyone wanting to try this:

```python

from pycatia import catia
from pycatia.drafting_interfaces.drawing_document import DrawingDocument

ca = catia()

documents = ca.documents
all_sheets = []

for document in documents:
    if document.full_name.endswith("CATDrawing"):
        drawing_document = DrawingDocument(document.com_object)
        for sheet in drawing_document.drawing_root.sheets:
            all_sheets.append(sheet)

current_sheet = all_sheets[0]

background_view = current_sheet.views.get_item_by_name('Background View')
background_view.activate()

tables = background_view.tables

test_table = tables.add(100, 100, 7, 9, 6, 20)
```

ab0v3g4me, 2021-04-28:
Appreciate the answer!! Thanks a lot!! And indeed when i lower the font size all is well :)!

closed 2021-04-29 by evereux