evereux/pycatia #92: Export geometry with custom Axis System

j-bau, 2021-06-16:
Hello,
I want to export my geometry with respect to a custom coordinate axis system. In the export case the default is set to the global axis system. Thus my exported data is placed and rotated in a wrong way.
The help to export VRML for instance (https://help.solidworks.com/2017/english/SolidWorks/sldworks/HIDD_EXPORT_OPTIONS_VRML.htm) states that a "Output coordinate system" can be picked. I cant find the option in your export_data(...) function call. Am I missing something or can you give me a hint?

evereux, 2021-06-20:
From what I've read the CATIA Automation API that pycatia uses doesn't provide such a tool.

You would have to use a transformation matrix to translate the co-ordinates from the absolute co-ordinate system to the axis co-ordinate system you'd wish to use.

It would be nice to incorporate such a thing into pycatia if anyone would like to take that on. Should be quite straight forward if already familiar with the math involved.

evereux, 2021-06-20:
You can retrieve information about the axis system you with to measure from using something like this:

```python

from pycatia import catia

caa = catia()
documents = caa.documents

document = caa.active_document
spa_workbench = document.spa_workbench()
part = document.part

# get axis system with name "MyDatum"
my_axis = part.axis_systems.get_item_by_name('MyDatum')
my_axis_reference = part.create_reference_from_object(my_axis)

measurable = spa_workbench.get_measurable(my_axis_reference)
measurable_axis = measurable.get_axis_system()
# example output: (100.0, 100.0, 100.0, 1.0, 0.0, 0.0, 0.0, -0.3420201433256687, 0.9396926207859084, 0.0, -0.9396926207859084, -0.3420201433256687)
# see https://github.com/evereux/pycatia/blob/cff309fe2b4802ff2b2c5c984f8064747f81065d/pycatia/space_analyses_interfaces/measurable.py#L267
```

j-bau, 2021-06-21:
Thanks for your input. In the meantime I found an other way to do it. Maybe it is helpful to mitigate all the transformation implementations by hand. 

The Idea behind it is to use the already implemented axis-to-axis transformation. With this option a custom axis can be transformed to the origin axis. 

```python
from pycatia import catia

caa = catia()
documents = caa.documents

document = caa.active_document
spa_workbench = document.spa_workbench()
part = document.part

axis_systems = part.axis_systems
axis1 = axis_systems.get_item_by_name('Absolute Axis System')
axis_ref1 = part.create_reference_from_object(axis1)
axis2 = axis_systems.get_item_by_index(2) # just pick first custom coordinate system
axis_ref2 = part.create_reference_from_object(axis2)

current_hb = part.hybrid_bodies.item(1) # just pick the first item for this test
current_hs_list = current_hb.hybrid_shapes.get_item_names()
# iterate throu hybrid shapes and create an axis-to-axis element
for i in range(len(current_hs_list)):
    current_hs = current_hb.hybrid_shapes.get_item_by_name(current_hs_list[i])
    a2a = part.hybrid_shape_factory.add_new_axis_to_axis(current_hs, axis_ref2, axis_ref1)
    current_hb.append_hybrid_shape(a2a)
# update catia project
part.update()
```

evereux, 2022-04-19:
Closing as OPs issue has been resolved.

closed 2022-04-19 by evereux