evereux/pycatia #93: [BUG]pywintypes.com_error when executing example_003

CarrotC, 2021-07-12:
**Describe the bug**
[2021-07-12 19:14:46,962] INFO in documents: Opening document "D:\data\CADModels\catia\BHHelicopter\M24.CATPart".
[2021-07-12 19:14:47,413] WARNING in document: This method may be deprecated in future versions.
Traceback (most recent call last):
  File "D:/workspace/python/STLCompare/PointsFromCATPart.py", line 30, in <module>
    selected = document.search_for_items(['Point'])
  File "D:\workspace\python\STLCompare\venv\lib\site-packages\pycatia\in_interfaces\document.py", line 844, in search_for_items
    selection.Search(query_string)
  File "<COMObject <unknown>>", line 2, in Search
pywintypes.com_error: (-2147352567, '发生意外。', (0, 'CATIASelection', '方法 Search 失败', None, 0, -2147467259), None)

**To Reproduce**
run the example_003 code with a CATPart file

**Expected behavior**
get a csv file of the points of the CATPart

**Desktop (please complete the following information):**
 - OS: Windows 10
 - CATIA V5 version:  V5-6R2017
 - pycatia version: pycatia 0.5.0

**Additional context**
Besides, I want to ask whether this example code is to get points by sampling the CAD model to a point cloud or get the vertex.

Tian-Jionglu, 2021-07-13:
Hi, this is caused by you Chinese environment.
This example use search method, and only valid for name `Point`.

CarrotC, 2021-07-13:
> Hi, this is caused by you Chinese environment.
> This example use search method, and only valid for name `Point`.
Hi, there is no Chinese word in my path, or you mean I should change the language of my computer to English?  
I'm not familiar with CATIA. is it none if I search name `Point` in a Cylinder CATPart?

evereux, 2021-07-13:
If Points were used to define the Cylinder and they are named Point in your tree the script example should find them.

If you want to find the centre points at the ends of a Cylinder solid, and those Points aren't used for construction, you'll need to tackle this differently.

CarrotC, 2021-07-13:
@evereux  Get it, thanks. 
Do you know how to deal with the pywintypes.com_error reported?

evereux, 2021-07-13:
Can you provide a picture of your part tree.

evereux, 2021-07-13:
For example
[image]

CarrotC, 2021-07-14:
I tested several CATPart including a Cylinder without Points, and the CATPart produced by example_005. Their part trees are 
[image]
[image]
If this error caused by the CATPart, could you upload an example CATPart that runs well?

CarrotC, 2021-07-14:
This problem is solved after I changed the Interface Language to English.
And I understand the concept of Point in CATIA with your explanation.
Thanks a lot.

closed 2021-07-14 by CarrotC