evereux/pycatia #94: [FEATURE REQUEST] How to set current shape in boolean operations

CarrotC, 2021-07-20:
There is a VBA  example of boolean operation in pycatia.part_interfaces.intersect.Intersect(com_object), showing that
```
‘ Make Pad1 the current shape
CATIA.ActiveDocument.Part.CurrentShape = Pad1
‘
‘ Create the intersection between Pad1 and Body2
Set NewShape = shapeFactory.AddNewIntersect(Body2)
```
But there is no current_shape attribute in pycatia.mec_mod_interfaces.part.
For example, I add several bodies and want to intersect body.3 and body.2. Without setting the current shape, the body.3 will be intersected to the last body I added.
How can I set the current shape in boolean operation?

evereux, 2021-07-25:
This example has a CATPart with two bodies called Body.Cube.1 and Body.Cube.2.

Body.Cube.2 is intersected with Body.Cube.1.

The current_shape is I'm guessing the in_work_object which you can set as shown below.

```python

from pycatia import catia

caa = catia()
documents = caa.documents

document = caa.active_document
part = document.part
bodies = part.bodies

body_cube_1 = bodies.get_item_by_name('Body.Cube.1')
body_cube_2 = bodies.get_item_by_name('Body.Cube.2')

part.in_work_object = body_cube_1

shapes_body_cube_1 = body_cube_1.shapes
hybrid_bodies_cube_1 = body_cube_1.hybrid_bodies
hybrid_shapes_cube_1 = body_cube_1.hybrid_shapes

shape_factory = part.shape_factory

shape_factory.add_new_add(body_cube_2)
```

evereux, 2021-07-25:
This was a good question that's not easy to solve reading the documentation. I'd like to add this as an example but I encourage others to contribute if they can. So, if you're willing you're more than welcome to do that, using the existing examples as a template.

CarrotC, 2021-07-26:
Thank you. I will try to do it.

evereux, 2022-10-08:
I've now added the example so will close this issue.

closed 2022-10-08 by evereux