evereux/pycatia #97: [BUG] GSD Disassemble surf with add_new_datums command error

bhulele, 2021-08-07:
**Describe the bug**
I'm trying to split the shape in the attached file(surf.CATPart) into its two domains programatically with  the add_new_datums command, but the command "hybrid_body.append_hybrid_shape(oDomains[ibound])"  fails because "AttributeError:  addNewDatums.com_object"
It is the result of a split operation that gives multiple domains. I have used pycatia quite a bit.  Any ideas how to get to the correct script of this split seperately ?

I have searched a lot with Disassemble surf with VBA, the attached script file(VBProjection1.catvba) can successfully disassemble surf.      [http://www.coe.org/p/fo/et/thread=26745#]

My code is attached blow(surf_disassemble.py)
Thank you very much.
[catia_VBA_py_files.zip](https://github.com/evereux/pycatia/files/6948886/catia_VBA_py_files.zip)

**Expected behavior**
I want to disassemable surf to sub elements.

**Desktop :**
 - OS:  Windows 10 
 - CATIA V5 version: V5R20
 - pycatia version: pycatia 0.5.0

**error message**

  File "C:\Users\lele\Desktop\surf_splite.py", line 42, in <module>
    hybrid_body.append_hybrid_shape(oDomains[ibound])

  File "e:\anaconda3\envs\pycatia\lib\site-packages\pycatia\mec_mod_interfaces\hybrid_body.py", line 209, in append_hybrid_shape
    return self.hybrid_body.AppendHybridShape(i_hybrid_shape.com_object)

  File "e:\anaconda3\envs\pycatia\lib\site-packages\win32com\client\dynamic.py", line 527, in __getattr__
    raise AttributeError("%s.%s" % (self._username_, attr))

AttributeError: AddNewDatums.com_object

evereux, 2021-08-08:
This had me stumped for awhile! :-)

The problem lay with class method `add_new_datums`. What it was doing was returning a `tuple()` of `com_objects` (CATIA based objects if you likle) in lieu of python Class based objects.

So, when you were doing` hybrid_body.append_hybrid_shape(oDomains[ibound])` pycatia was getting the `com_object` and not the expected python Class object.

I've submitted a fix and pushed that to the master branch.

I'll make a release in the next couple of days.

btw, not sure what your intent was but if you wanted all the datums you should just pass the total number. Not num -1.

```python

for ibound in range(num):
    hybrid_body.append_hybrid_shape(oDomains[ibound])
    part.update()
```

>  I have used pycatia quite a bit.

That's good to hear. Never quite sure if people use it in anger or just play with it now and then.

Cheers

evereux, 2021-08-08:
ps thanks for the stripped down example code and part. It helps me a great deal when people do this.

evereux, 2021-08-09:
I have released the new version 0.5.1 that fixes this issue.

bhulele, 2021-08-09:
Thank you for you suggestion, I should pass the total number.
pycatia is very convenient and  useful.
I have update the new version 0.5.1 and get the correct result. 
Thank you very much again.

closed 2021-08-09 by bhulele